prusa3d/PrusaSlicer #13188: Only one perimeter on topmost surface: doesn't work in multimaterial setups

k1-801, 2024-07-31:
### Description of the bug

[image]

When the "only one perimeter on top surfaces" is set to "only topmost" in a multi-material print, it only affects the last part of the multimaterial object. It is clearly seen on the screenshot. I only tested it with different materials being different objects, did not test it with paint-on colors yet.

Setting a modifier for "only topmost" for the object that wasn't affected by the global setting also does nothing.

### Project file & How to reproduce

[Topmost error.zip](https://github.com/user-attachments/files/16448031/Topmost.error.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 10

### Printer model

Creality K1 Max with ERCF

k1-801, 2024-08-01:
UPD: the issue seems to be inconsistent, did not happen on the other sample plate in the same project file.

u89djt, 2024-08-01:
Did you post the right 3mf file? Both objects slice with a single perimeter on top for me in 2.8 windows 11

u89djt, 2024-08-01:
This is the ordinary pain around confusion and bugs - keep it up, you'll find something.

closed 2026-09-29 by prusaslicer-bot[bot]