prusa3d/PrusaSlicer #13191: Printed Gear ALWAYS has a "rib" created relative to the hub recessed area.  All prusa slicer versions and ALL MK3 and Mk4 printers

kevinmichaeldudley, 2024-08-01:
### Description of the bug

All circular DISC objects with a recess in the dish/disc cause a "rib" to be created AT the same level as the transition from the "recessed area flat" to the fillet.  This has occurred for YEARS and I have reported before but no response.
I print gears but ANY disc shape with a recessed hub can cause it.  But, LARGER DIAMETER AND DEEPER RECESS AND SHARPER FILLET cause it worse.

[image]
 

### Project file & How to reproduce

[Prusa slicer bug on gears.zip](https://github.com/user-attachments/files/16462090/Prusa.slicer.bug.on.gears.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Happens all versions from 2.3 and up, latest is 2.8.0

### Operating system

Windows 11

### Printer model

Mk3 Mk3+ and Mk4.  8 different printers.

u89djt, 2024-08-02:
You've been stuck for years, so presumably you'll welcome a fresh take on it from a fellow user.
First question: does it happen with all default parameters? I see you've got parameter changes and custom g-code. The defaults are chosen to work well together. Maybe you've asked for too much plastic to be squished together in circumstances that lead to the ridge. The gcode you've posted confirms that the extrusion paths per se don't indicate a ridge. You've concluded that there's a bug with the printers. I've never seen anything like this on a MK3S and an XL, but I haven't tried your parameters.
If I switch to MK3S (not plus) and keep all the parameter changes but not the custom gcode, I get this:
[image]
There's no sign of a ridge. The top hypotheses might be that it doesn't happen with a wedge cut out of the gear, or that it's a result of the custom gcode. Cutting out the wedge seems unlikely to be important. You've said the problem worsens at larger diameters, so a straight edge - an infinite radius - would show it. You could make a short rack with the same context and see if it happens to that. I haven't thought of any reason why a closed disc could cause a glitch yet. If a closed loop extrusion using an arc substitution was involved, that could be worth looking at, but these are gear teeth. It would have to be an arc in the wrong place causing plastic squish laterally to be sustained all the way out to the surface. And arc substitutions are recent - you're looking at years.
Say and ask whatever comes to mind.

u89djt, 2024-08-02:
@neophyl you'll see in the gcode that there's no comparable infill transition in this model, and in the photo that there is a conspicuously protruding line that seems to be comparable with an extrusion width, rather than a minor, mostly cosmetic ripple. Note the way the illumination catches it.

neophyl, 2024-08-02:
I does look more extreme than any usual hull line problem.  Only time I have seen something as bad is when the stepper on Z has failed to move it up a full layer on a printer.  Given this is on multiple printers though it would tend to rule that out.
Have you tried setting external perimeters first ?  With this geometry there should be no problem doing so as a test to see if it makes any difference.

u89djt, 2024-08-02:
@kevinmichaeldudley I'm printing the whole disc for the avoidance of any doubt. To reduce print time/material consumption, I cropped it up to 9mm above the base so there are a few layers of infill before the solid surfaces, and it'll stop just above the inflection point between the inside and outside fillets behind the teeth. I'll collect that in the morning.

u89djt, 2024-08-03:
@kevinmichaeldudley no sign of disturbance around the start of the fillet for me. I have an aggressive first layer offset so there's an elephant's foot. Your rib would be expected around 3.802mm up where the inner surface and fillet start (as opposed to the 12.802 in your original). If you can execute a print on a MK3S that has the ridge and send me that exact 3mf file, I'll print the whole thing. Do me a favour and crop it down to the minimum height that causes it so I'm not throwing unnecessary plastic at it.
[image]
Here's the 3mf of the stunted ring:
[shorter.zip](https://github.com/user-attachments/files/16478849/shorter.zip)
And here's the wedge I printed previously
[full height wedge K1050 SB-13 80T Standard Offset V6 MK4.zip](https://github.com/user-attachments/files/16478850/full.height.wedge.K1050.SB-13.80T.Standard.Offset.V6.MK4.zip)

u89djt, 2024-08-03:
Yeah, post the smallest thing that definitely exhibits the problem. It might take some iterations to find the problem.
[image]

u89djt, 2024-08-03:
Assuming that's yours, what did you do differently in item 125739486192 ? That doesn't seem to have the ridge.

u89djt, 2024-08-03:
@kevinmichaeldudley see if you can identify a relatively small object that exhibits the problem on a MK3S and we can compare results. Do you have any objects of this general form that don't show a ridge? I've deleted previous comments as they didn't really add anything.

kevinmichaeldudley, 2024-08-04:
Printing wedges: will always print perfectly, it only occurs on full “circle” prints.

MK3: I only have “new” MK4’s now, all my MK3’s burned up with the house in June ;>(

Smaller diameters: The issue NEVER shows on diameters much below 75 mm

Your slice:  I’ll print the same 3.8 mm up to see if it does not show for me. (not surprised as “thickness” does seem to have a part)

Your comment: Assuming that's yours, what did you do differently in item 125739486192 ? That doesn't seem to have the ridge.

Yes that is mine.  Just MUCH larger fillet (blend towards the inner hub).  ½ wide gear vs .75 in  (seems to play a part in severity).

 

The gear model I sent you was 4x worse than when printed on the MK3.  On the MK3 it is just enough to show a mark.

 

Here is another gear, thinner gear 116T which has the more typical that I see.  Just enough to be a “blemish” but not enough to actually impact part service.  I just hold my breath and no customer has complained.

 

I’ll work up a thinner sample such as this gear to send a model.

 

I have studied the gcode as it is a copy layer after layer (on the perimeters at least) and I see no variance.

FYI: it is always slightly, slightly shifted, higher on one side than the other but ALWAYS returns back to normal position and finish.

 

I appreciate the attention and help.  I will put more together and send to you.  Could be a few days.

Thanks so much!  This drives me crazy.  I ran TWO of the large gears ($10.00 gone) before I noticed the “witness mark” was MUCH more severe  on the MK4.

 

Also, I did flatten the fillet AND ran as “structural” on the MK4 and got a near perfect part.

 

I will be in touch soon.

 

20240803_100011.jpg (view on web) <https://github.com/user-attachments/assets/84129cbd-55e9-423c-9a6a-13a1dce308c3>

u89djt, 2024-08-04:
No worries - this is a great way to learn stuff before it gets in the way of my own projects :)
It would be good to see what's happening physically during deposition of the ridge layers. Maybe stop the print when the ridge has been layed down - or is it squished by the layer above?
It would be handy for someone with a MK4 to chime in, really. I haven't had any plausible ideas.

DaGwhite, 2024-12-09:
I have run into what I think is a related issue  off & on for a long time.  There's close to 30 pages of PrusaSlicer forum posts on the "bulge/"buldge" problem, which has been around for at least 6 years.

I ran into it again recently, with a very large defect in a very simple small model.  

[image]

In the hopes that it will help somebody finally resolve the problem, I did a bunch of tests and took a bunch of photos.  All the gory details are in an attached PDF, and I have assembled a zip file with the model, print files, etc.

[PrusaSlicer Bulge Bug.pdf](https://github.com/user-attachments/files/18055320/PrusaSlicer.Bulge.Bug.pdf)
[PS Bulge Test.zip](https://github.com/user-attachments/files/18055321/PS.Bulge.Test.zip)

u89djt, 2024-12-09:
@DaGwhite the 3mf files you provided don't contain the printer profile. If you can zip a project file that gives the problem for you - so that's the whole thing with the printer profile too, right as you're about to hit slice then export - I'd like to print it on my XL and MK3S to see it in person. Obviously I will be switching to the appropriate profiles. I want to see what your printer profile is like, and to be sure of the parameters of the print.

DaGwhite, 2024-12-09:
I wasn't sure which of the export options you need, but the bundle with "Physical printers" sounded most complete.  I've attached the resulting .ini file.

Thanks for digging into this.  If the bug appears on your printers as well, then it will be really helpful in chasing this down.  The reason it's been haunting folks all this time is that it is VERY sensitive to the specific model & print config.  In this case, I would never have encountered it if I hadn't needed to reduce the layer height for the fine thread I was printing.

Best of luck & keep us posted.

[PrusaSlicer_Bulge_Bug_config_bundle.zip](https://github.com/user-attachments/files/18062836/PrusaSlicer_Bulge_Bug_config_bundle.zip)

u89djt, 2024-12-09:
OK, I did something that meant your 3mf file didn't load your printer config. I'm not sure what I did wrong, but now the 3mfs you originally posted load the printer profile. Sorry to put you through the extra work, but maybe the ini files will include some important clue.
So let's be as meticulous as we can, because clearly I can goof here.
I'll print "PS Bulge Test (No Hole, Half).stl" on my XL in default profiles "0.1 DETAIL" and "0.2 SPEED" by importing the stl file into an empty slicer and we'll look at that and pick what to do next.
This is where it all slows down again, but I'm in favour of working out what's happening. You'll see that I didn't experience ridges in the prints I did above. We're trying to get a ridge on one of my prints, which would tell us something about the settings that lead to it.

DaGwhite, 2024-12-09:
The "PS Bulge Test (No Hole, Half).stl" does NOT exhibit the bulge when I print it.  Only the full model (with or without the hole) does.  I also included a STEP file of the model so folks can play with that to see what might be triggering the bulge behavior.

u89djt, 2024-12-09:
The dumb thing I did was not select open project after I had selected open models only from a previous issue - I so rarely open just the objects from a 3mf file (maybe never before that), it didn't register that the wrong radio button was selected despite it being right there in front of me.

u89djt, 2024-12-09:
Right, so that's the object that is active in the 3mf file you posted. I need to print one of the other objects.
To make absolutely sure we're looking at the right thing, please could you comment with a zip file of the stl of an object that causes a ridge for you, and a 3mf file or both 01 and 02 with just that object with something in the file name indicating whether or not a ridge appears so everyone will definitely be looking at and trying to print the same thing?

DaGwhite, 2024-12-09:
Sorry about the infil change.  I was experimenting with that, and ran out of time.  The original zip file was from a week or two back, and the full config was from today.  I don't think changing the infil made any difference to the bulge, but I don't recall for sure if I ever actually had time to try it.  I'd stick with whatever setting was in the original zip file to make sure if you can reproduce the defect.

The infil comments I made in the PDF relate to the wall curvature I saw in prints from both PrusaSlicer and Simplify3D.  I think that is a separate issue from the bulge.  As you say, it's probably an infil shrinkage issue with too thin a wall, inadequate cooling, or some combination.

u89djt, 2024-12-09:
There are two different infill regimes in the original zip.

u89djt, 2024-12-09:
Let's breathe for a couple of days. Here's a 3mf with a print profile I can run. You're welcome to set up parameters in this to illustrate what seems worth trying. Put other objects in, whatever you feel like. I'll then take a view on printing the whole project.
[battleofthebulgeXL.zip](https://github.com/user-attachments/files/18064871/battleofthebulgeXL.zip)

DaGwhite, 2024-12-09:
> There are two different infill regimes in the original zip.

I'm not sure how/where that's possible...  I'm not a particularly sophisticated user, and I certainly didn't do that intentionally...

I've got a print job running now, but may be able to take a look at your profile later today.  I have a lot of CAD work to do before I print my next big project, so I should be able to slip in some experiments here & there.   

I'll start by trying to get your profile working, and see if I can duplicate the problem.

u89djt, 2024-12-09:
Eh, none of us are as sophisticated as we'll be next week after working on stuff.
That profile won't work for you, per se, it's just a way of giving me something I can print without getting confused over versions of files and such. Do you know how to set per-object modifiers? Right click on an entry in the list and "add settings".
I'm not in a hurry - I'll get a notification if and when you post :) Hopefully other folk will be keeping an eye out, too. I'm just another user. When we start wasting time/space, it'll be good to get a nudge from an expert.

DaGwhite, 2024-12-10:
I took a look at the "battleofthebulgeXL.3mf" you provided.  As it was setup, I don't think it will produce the defect because you've got a 0.2mm layer height.  In theory, you should be able to use your printer & filament settings, but the Print Settings needs to be "MG M2E Baseline 0.1 mm Layers (NG Scope Rings)" to duplicate the configuration that produces the defect.  

I made a copy of "battleofthebulgeXL.3mf", and changed it to use my printer and filament settings, but when I do that, your "battleofthebulgeXL.3mf (0.20mm SPEED @XLIS 0.4)" Print Settings disappears.  I'm not sure what is going on...

u89djt, 2024-12-10:
Don't worry, that's to be expected. When I said it wouldn't work for you per se, I suppose I meant that this is what would happen were you to try to use it yourself. When I get a chance, I'll have a go at composing a good explanation. In the meantime, have a look around the prusa knowledgebase. Comment about whether or not that clears up the confusion when you've looked around.

u89djt, 2024-12-10:
I should be candid: I'm quite sure the slicer isn't at fault; that there's something erroneous in the print profile you're using, or something out of adjustment or faulty on your printer. If I'm wrong and there *is* something at fault in the slicer, then I want to know about it. I haven't been able to persuade either of my printers to create ridges of the kind you and the original poster depict.

DaGwhite, 2024-12-10:
I understand your skepticism.  I and many others have gone over our printers & slicer settings with a fine tooth comb trying to find a fix.

If you get bored sometime, try wading through the 30 pages of user forum postings I linked to in the PDF.  The issue was first identified about 7 years ago, and users have tried all the new releases of PS since then in the hopes it would go away.  After all sorts of experiments to pin it down to a variety of slicer settings, nobody has ever been able to identify ANY setting changes in PrusaSlicer to make the problem disappear.   

I've had plenty of prints where switching to Simplify 3D makes the issue go away, and others have switched to Cura and found their prints to be suddenly bulge-free.  This is also with a wide range of printers.  If it's not Prusa Slicer, then I don't see how using a different slicer would fix it.  Same for it being a printer issue.  The common element in all the forum postings is PrusaSlicer.

At least in the latest case, the problem appears to be triggered by the specific geometry of the model, AND by using 0.1 mm layer height.  I've had plenty of prints where the problem occurs at 0.2 mm layer height as well, but always with larger, more complex models, and generally not as bad as what I see with this one.  That's why I posted the files.  If you or someone else can see the problem on their system, then we may be able to drill down and find the cause.  If you can't recreate the problem, that also tells us something, and maybe we can ferret out why and find a fix that way. 

One thing I can try is to use the PS default print settings, but set it to 0.1mm layer height and see if that still creates a bulge.  I already know the perimeter generator doesn't matter.

DaGwhite, 2024-12-10:
Test complete:  I went in, and created a "PrusaSlicer Defaults" Print Setting profile.  Other than the 0.1 mm layer height, I went through **every** menu and reset everything to use the default settings.  The result was a defect just like before.  If anything, it's actually slightly worse.  The "bulge" on one end is actually on the order of a full extrusion width, and the print is sagging there where it's printing out in space.

[image]

Next, I will try printing with 0.15  & 0.2 mm layer heights.

u89djt, 2024-12-11:
It sounds like the gcode from different slicers for the same object need to be examined in detail?

u89djt, 2024-12-11:
You don't seem to be using one of the M2E profiles provided in the slicer. What happens if you do?
[image]

DaGwhite, 2024-12-11:
> I haven't found myself bored while pursuing a 3D printing issue yet. Does that step appear in the gcode? If not, then your printer is not executing the gcode correctly.

I haven't dug into the gcode (yet).  One reason I've been excited about this model is that it's small & simple enough that the gcode should be a lot more manageable to wade through.  

When I first started using PrusaSlicer, they didn't have a MakerGear M2E profile.  Another M2E owner provided the one I'm using.  I can try the official one at some point.

Tests at 0.15 mm & 0.2 mm layer height both largely eliminated the defect.  They aren't perfect, but the "bulge" is MUCH smaller.  They are consistent with more typical "ghosting".  I will try to get pictures later today.  I need to see if I can come up with a measurement I can use to quantify the size of the defects.  

Thanks for the feedback!

DaGwhite, 2024-12-12:
I've been busy fighting other fires, but here's a few updates:

I installed the official MakerGear M2 profile, and ran the model with the PS default print settings, but 0.1 mm layer height.  That had the bad defects, but the overall print quality was also noticeably worse.  I suspect the speed settings may be more aggressive than the user developed profile I have been using.  The official profile has a much nicer model of the bed, so I will probably go through it at some point and chase down and fix the differences that are degrading the print quality.

I also took a close look at the preview in PrusaSlicer, and there is no sign of the defect.  I didn't trust the preview coming from the same possible source of the defect, so I also used the gcode viewer in Simplify3D.  There is no defect in that preview either.

That points to a printer problem, but there are several things about that conclusion that make no sense to me.  

1)  This issue has been reported on multiple different brands and models of printers.   If it's a printer issue, it has to be something common to a bunch of them.   My M2E dates back to 2017, and is running Marlin 1.0.2  (January 2015) on a Rambo 1.4 board.  That is a very common configuration, and it's possible that the other users with this "bug" are running that same hardware/firmware.   I have a beta version of Marlin 2.0 for my machine, but when I installed it, all my motors ran backwards...  I reverted to my old firmware, and haven't had time to track down & fix the motor issue.

2)  The gcode commands are primarily just a series of position & extruder operations.  I'm seeing a very significant shrinkage of the walls as the print progresses, and then the extruder suddenly comes to its senses at the transition level, and goes back to printing proper dimensions.   I don't see how the stepper motors can slowly send the bed to different locations as the height changes.  If we believe the previews, the gcode is telling the XY steppers to trace out exactly the same perimeter layer after layer.  Steppers shouldn't drift...  The printer firmware shouldn't be capable of cleverly adjusting both X & Y position to taper the walls.

3)  One possibility is that the shrinkage is a pathological model dependent thermal effect, and the hardware & software are actually OK.  The print time of a layer changes dramatically when the print reaches the level where the defect occurs.  That means the layers cool differently.  If layer-time dependent fan cooling is enabled, that could also trigger a large change in the shrinkage of the print in certain areas.  The defect appears to be slightly different depending on which side of the model is facing the fan.  

4)  Why is the defect dependent on layer height?   It's quite small when I print at 0.15 and 0.20 mm height, but very pronounced at 0.1 mm height.  It's like it's falling off a cliff, rather than just getting linearly worse as the layers get thinner. This may tie into the thermal/shrinkage angle.  At 0.1 mm layer height, the "cooling" of the nozzle due to the material being extruded is reduced.  At 0.2 mm layer height, molten plastic is getting run out of the nozzle much faster, keeping the nozzle and extrusion temperature down.  It's possible the shrinkage of the material is non-linear, and running with thin layers is pushing the nozzle temperature over a threshold where bad things happen.  The temperature sensor is not right at the tip of the nozzle, so the actual extrusion temperature could vary a bit due to flow rate (i.e. layer height).  I've run into the "bulge" problem when printing at 0.2 mm but with very different models.  Whatever is going on is highly model dependent.  On the gear design at the start of this thread, the layer print time will drop dramatically when you get above the height of the main disk, and are only printing the rim and the hub.  There are actually three regimes: 1) Printing a large volume in the lower portion, 2) printing the top layers where the transition occurs, and 3) printing the upper portion where there is a LOT less material being extruded.

I need to ponder this some more before I try additional experiments.   I'm leaning towards items 3 & 4 being involved.  That would mean that carefully adjusting temperatures, speeds and cooling should fix it.   For example, I could try dropping the temperature setting on the 0.1 mm layer model and see if the defect gets smaller.  If the extrusion nozzle temperature is too high due to low flow, lowering the print speed could actually make things worse...

u89djt, 2024-12-12:
OK so at this point, we know the slicer software itself isn't at fault, so the exploration belongs somewhere else. The off-brand profiles included with the slicer are generally developed by expert users of the third party equipment. The specific printer you own - in concert with the filament you're using - fares worse when that profile is used instead of the one you sourced elsewhere historically. I think that suggests that the community around your printer may have learned things that you need to find? There's also the possibility that there may be luck of the draw involved in specific components. I therefore wish you luck!

DaGwhite, 2024-12-12:
Not so fast.  The fact that the preview shows that the extruder is being commanded to go through the correct mechanical motions doesn't mean PrusaSlicer is off the hook.  If the problem is excess shrinkage, that is a speed/temperature/cooling issue, which the preview doesn't show.  Those are all parameters that other slicers will treat differently, which may be why the "fix" is to use Cura or Simplify 3D.  

As requested, I have now demonstrated that I can reproduce the defect using PrusaSlicer's default Printer Settings, AND their recommended Printer Settings for my machine.  The only thing I haven't set to their defaults is my filament profile, because I'm using a PLA Plus material that requires higher temperatures.  The material the OP's gears were printed with looks to be very different from what I'm using, and yet the problem remains ON HIS PRUSA PRINTERS ("Mk3 Mk3+ and Mk4. 8 different printers.").

Have you tried to print my model yet?  Before you blame my printer, could you see if the model prints OK on your system?  In addition to the original poster, I believe there are other people who have Prusa printers who have reported the problem on the user forum.

In the meantime, I think I will try a print at a lower temperature., but I'm not running super hot as it is.  I can also do some research on what affects shrinkage.

DaGwhite, 2024-12-12:
Dropping the temperatures 5C didn't make an appreciable difference. In looking over the various models, switching to the PS M2 profile did help a little, and I will have to investigate that.  The drop in print quality suggests that they are printing faster.  I prefer quality over speed, and my profile is probably printing slower, which gives the PS Auto cooling more time to shrink things...

I checked, and with the PS M2 profile, a print is supposed to take 31 minutes.  With my profile, it says 34 minutes, so the difference isn't very big.

Next, I will do a print with:  

"Print Settings" set to PS defaults except layer height = 0.1 mm
"Filament Settings - Cooling" set to the PS PLA defaults
"Printer Settings" set to the PS MakerGear M2 profile

I'm also going to rotate the model 90 degrees and set the seam to be at the rear.   That will put the seam in the middle of the arc on one end.  PS is currently parking it at a corner, where it interferes with examining one of the defects.

DaGwhite, 2024-12-12:
Using PrusaSlicer's default cooling configuration produced no improvement.

I now have a print on the machine with cooling disabled entirely...

DaGwhite, 2024-12-13:
Well, turning off PrusaSlicer's "auto cooling" made the biggest improvement I've seen so far.  There is still a defect at the height of the transition from the main body of the model to the four legs, but it is MUCH smaller.

The overall print quality has deteriorated significantly with all of the default settings I'm currently using.  I want to go back to my original model, and see what happens to it when I disable "auto cooling".

There's still some tests I'd like to repeat and refine, but the "bulge" appears to be largely the fault of the auto cooling algorithm in Prusa Slicer.   It's hard to say exactly what it's up to that is causing the problem, but the most likely thing is overly aggressive cooling, which produces excessive shrinking.  The bulge is actually produced when the geometry changes enough at a transition in the model to eliminate the shrinkage, at which point the model appears to expand.  It's actually just going back to where it is no longer shrunk excessively.

u89djt, 2024-12-13:
Here's 0.1 layer height default settings on the XL. There's a line on one of the uprights that I can maybe just feel with a finger nail.
[image]
And here's your project with just the printer profile changed to the XL
[image]
Each extrusion shrinks and pulls at the plastic already laid, and if that's above the glass transition temperature it will creep permanently. You've got four perimeters and dense infill, so I guess that's why it shrinks so much.
I suggest that the bulge is the extrusion escaping down the side of the layer below that has shrunk back out of its way. Easier to go that way than to squish along between the nozzle and the plastic in place. The layer above might be disturbed inward by the bulge accentuating the effect.
The line on my defaults print seems reminiscent of the benchy line, @neophyl ?
https://help.prusa3d.com/article/the-benchy-hull-line_124745
The explanation offered for the benchy line does seem to cover the issues around your bulge and offset when taken to an extreme by number of perimeters, density of infill and contrast between print times of layers.

DaGwhite, 2024-12-14:
Thank you for running those tests.  You are getting the defect when you run my Print and Filament settings with your XL printer profile, but it's almost invisible with **your** print and filament profiles.  That means there are settings in those sections that make a BIG improvement.  I don't understand what settings those might be, because I don't see the same improvement when I run your Print & Filament settings with my Printer profile.   When you didn't see the defect, were you running all the same settings that are in the "battleofthebulgeXL.3mf" you posted?

I've been re-running some experiments, and getting results that are a bit inconsistent with what I had before.  I need to go back through things carefully and see if I missed some settings/changes.  One issue is that my printer is in my basement, and it's a lot colder today than when I ran a lot of the earlier tests.  I'm not sure, but given that over-cooling seems to be a critical part of the problem, that may be making it a bit worse.  I definitely see a big improvement when I turn off "auto cooling". 

I spent some time today going through the PrusaSlicer MakerGear M2 printer profile, and I've identified all the differences between that and my older M2-user supplied profile.  They are very similar in most respects, and I'm not sure why the print quality drops a bit when I use the "official" version.

I had seen references to the benchy line issue, and suspected it might be related.  It's no where as severe as what I've been seeing with this particular model.  I checked out the link you provided, and they mention the cool basement issue.  Time to turn on the heat...

Your comment about the glass temperature got me thinking about cooling in a bit more detail.  I think some cooling would be useful, but if over cooling is bad and causes excess shrinkage, then dropping things just below the glass temperature (and no further) might be good.  I haven't investigated which cooling settings are active when "auto cooling" is off.  At the moment, the best results are when I have no cooling at all.  The only other option is to have the fan on all the time, but again, I'm not sure which of the other settings are in play in that mode.

u89djt, 2024-12-14:
Sorry, I failed to zip and post project files for the two prints! I spend half my time pushing people into posting the project. Colour me chastened.
This is using as much of your profile info as I felt comfortable letting loose on my printer, so that's just selecting the XL in the dropdown Printer list:
[001 241213 PS Bulge Test 0r1 mm Layers.zip](https://github.com/user-attachments/files/18135256/001.241213.PS.Bulge.Test.0r1.mm.Layers.zip)
And this is that file after selecting the XL 0.1mm default settings in the Print Settings dropdown:
[002 241213 01 fast detail xl PS Bulge Test 0r1 mm Layers.zip](https://github.com/user-attachments/files/18135258/002.241213.01.fast.detail.xl.PS.Bulge.Test.0r1.mm.Layers.zip)
If I remember correctly, the battleofthebulge file was an empty default XL 0.1 project with the source mesh inserted and cropped a bit to match yours. I will have done that to be sure of not having any of your settings in the project. Having concluded that it isn't a slicer bug - I think we agree that it isn't a bug at this point? - I tidied up and moved on.
So, selfishly, since I've now seen that I can suffer from the physics, I'm more motivated to dig. I've printed dozens of benchies and not had a benchy line that bothered me, so that's got to be luck of the draw in terms of equipment and where I tend to put it.
I assume at least one person at prusa is getting notifications about this thread in their issues system, and we're talking at length about something that isn't a slicer bug, so we're nibbling at their attention.
The knowledgebase article on the benchy line reminds us that a lot of actual expert time has been spent searching for good extrusion practices. But, they presumably would appreciate any new mitigation concepts. I think I would need to do a substantial job of reading and experimentation before I could be heading for that territory.
@neophyl brought up the benchy line early in the thread.

DaGwhite, 2024-12-14:
There is clearly a lot going on here.  I'm not sure that it qualifies as a "big", but the fact that their "auto cooling" makes the problem much worse suggests there is work to be done there.

It also occurred to me that the exact brand & type of filament could be important as well.  I use the 3D Fuel Pro PLA+ because it is significantly stronger than vanilla PLA, but it's thermal characteristics (like shrinkage) could be different enough from vanilla PLA to be part of why my setup is producing very large defects.  I hope it's not that, 'cause I just bought 4 spools on a Black Friday sale...

I've come up with a way of quantifying the amount of defect on my models.   I use a caliper to measure across the corners where the defect is located.  I take one measurement just below the defect where the shrinkage is at a maximum:

[image]

And then one just above the defect transition to measure the Post-defect size:

[image]

Half the difference gives me the average defect size along that diagonal, and then I average the results with the other diagonal.  With my original setup, I get a defect size of ~ 0.33 mm.  If I run the same file with "auto cooling" off (no cooling at all), it's about 0.18 mm.  With "auto cooling" on, the defect size shrinks as I increase the layer height, but it's still pretty bad even at 0.2 mm (defect size is 0.22 mm).

I did a lot better with the "battleofthebulgeXL.3mf" profile you posted earlier at thicker layers, and one of my projects for today is to track down which parameters are causing that.  I just ran that, and the overall print quality looks awful, but the defect is much smaller (haven't had time to measure it yet).  It's clear from your tests that there are some settings beyond turning "auto cooling" off that can improve things significantly.

u89djt, 2024-12-14:
When you tried the profile for your printer built into the slicer, did you use all default settings for one of the profiles? Numbers of perimeters, infill, speeds &c.?

DaGwhite, 2024-12-14:
> When you tried the profile for your printer built into the slicer, did you use all default settings for one of the profiles? Numbers of perimeters, infill, speeds &c.?

Yes.  A lot of the print quality issues are clearly the result of a coarse infil, and too few layers, especially on top.  I can see the infil pattern very clearly.

u89djt, 2024-12-14:
Does it shrink?

DaGwhite, 2024-12-14:
Not as much as my original settings.  I want to  go through it carefully and see if I can keep the low shrinkage but with acceptable quality.  Your example from yesterday looks really nice, and I want to take a look at what settings you used to get minimal defects with good quality.  I assume they are all the defaults.

u89djt, 2024-12-14:
They're the defaults for my printer, so they may be quite wrong for yours. You're asking for different physics to work. I would guess that the biggest difference could be cooling.
Cooling hard while extruding will cause most of the shrinkage to happen quickly, so as the extrusion progresses, the next hot bit will influence what came before less. It will tend to stay where it's put.
Try printing very, very slowly with maximum cooling running.
I don't think we're supposed to be pursuing physics in the slicer bug report system. The slicer does what you ask it to, including the auto cooling.

u89djt, 2024-12-14:
Is this what your part fan looks like? 
[image]

DaGwhite, 2024-12-14:
> They're the defaults for my printer, so they may be quite wrong for yours. You're asking for different physics to work. I would guess that the biggest difference could be cooling. Cooling hard while extruding will cause most of the shrinkage to happen quickly, so as the extrusion progresses, the next hot bit will influence what came before less. It will tend to stay where it's put. Try printing very, very slowly with maximum cooling running. I don't know how good the part fan is on your printer. I don't think we're supposed to be pursuing physics in the slicer bug report system. The slicer does what you ask it to, including the auto cooling.

If "auto cooling" makes a print significantly worse, the algorithm needs work.  Given that I have no idea what it's doing behind the scenes, it is NOT doing what I asked it to.  I ASKED it to produce nice prints without having to mess with a lot of settings individually.  If we can figure out what triggers the problem, maybe they can add an additional parameter or control so that it doesn't mess things up.  Given that it's all open source, there must be a description of what it's doing somewhere.  I don't have time to dig for that while I'm chasing the other issues.

The photo of the cooling set up is correct, with the fan blowing down from the front of the machine.  Now that we know cooling is a big part of the problem, I need to go back and revisit one change I made to the fan.  The original design has a plastic grill with a logo in the middle, which does two things; it makes it very loud, and it reduces the air flow significantly.  I replaced the grill with one that uses round wires, and an open middle.  That cut the sound level by 10 dB, but it also increased the air flow dramatically.  My plan was to print a fan duct that would steer that extra air more evenly around the print head, but that hasn't happened yet.  Other M2 users have made similar mods, and have never reported print problems.

When I've felt I was over-cooling, I've just reduced the maximum fan speed value in the cooling settings.  Until this adventure, it has never been a big issue, and I haven't had time to measure what setting would match the old grill air flow.  I print a lot of small parts, where the extra cooling usually helps so I don't have to slow prints down too much.  I want to document the print I ran last night with the "battleofthebulgeXL.3mf" profile, and then I will recalibrate the max fan setting to match the airflow of the original grill, but that will take a bit of time.  It's possible that once I do that, "auto cooling" will work better.

There is nothing I saw in the printer profile that talks about cooling performance, so you can't fix a machine specific fan issue there.  I took a look at the Prusa XL, and I can't tell how they do the cooling.  It looks like it must use a duct system of some sort.

u89djt, 2024-12-14:
OK, I'm going to move on. None of this is about the slicer. Best of luck.

DaGwhite, 2024-12-14:
I think "None of this is about the slicer" is a bit premature, but I have a lot of digging to do to figure that out, one way or another.   I've been taking fan air flow measurements all afternoon, and I need to print a new fan grill that will cut that back a bit without increasing the noise level.  I have a design that will do that, but it needs tweaking.

kevinmichaeldudley, 2024-12-15:
Hello, I am the original poster of the issue with the example of the gear.
I have followed your investigations, seemingly it is not settings related.
In my example I have found that printing perimeters FIRST removed 99.9% of any detectable “rib”.
Any other modifications of the model to print “less” would many times also eliminate the “rib”.
I printed my gear model on a Creatbot using a generated gcode from slicer, no rib.
I also had noticed that the “gear width” at the location of the rib was thinner, about 0.005”.
Opposite the rib, the gear was still the proper width.
The rib was ALWAYS only on ½ of the perimeter and ALWAYS the same orientation on the printer no matter the orientation of the model.
My feelings are the firmware is having some “accumulation” error which is occurring only under VERY limited model designs.  My gears all just happen to always display this “ribbing”.
As I stated, if I print perimeters FIRST the issues is 99.9% resolved.
I think some combination of number rounding / stepper half stepping / bed leveling (application) is causing the issue since I have had it for years on 8 different printers, Mk3’s and 4’s.  Also on every prusa slicer.
So, just sharing with you all since you are digging into this.  I have no answer other than the “Perimeters First” in my case.
On a side note, my company writes software for programming Lazer Machines.  We have discovered on some lazer machines the processors accumulate some error (we are supply rounded 3 place coordinates) and will error/stop.  In our software we have to “accumulate and store” the rounding value remainder and apply to the NEXT machine move to keep the machine happy.  Just saying…. Weird things with processors can occur.  FYI: A value of 1.0000 in a floating point processor is 0.999999997 or such.  Nothing is a PURE number!  We have had to programmatically account for this for 30 years, I know.
Hope this in some way helps,  Good luck!

kevinmichaeldudley, 2024-12-15:
Oh one last thought.  If someone could modify the firmware to "output the applied stepper driver steps" to a file or such,  I would think then software code could be ran to compare the "true stepper steps" to see if they "match" the request gcode moves.  Just looking for an error.
Thanks

DaGwhite, 2024-12-18:
Sorry to go silent for a bit.  I've changed the grill my printer fan so that it is cooling less aggressively.  I still get the best results with no fan at all.  I have determined that the problem can be reduced with adjustments to the slicer settings, but I have yet to find a configuration where it goes away to a reasonable degree, much less entirely.

Once I got the defects significantly smaller, I discovered that they are asymmetrical.  The defect is much larger on the front end than on the rear after I rotated the model to put the seam in the middle of the rear.  I think this may be the result of the "Y-shift" bug reported elsewhere:  [https://github.com/prusa3d/PrusaSlicer/issues/10423](url).  I may have to modify my model to rotate it and still have the seam end up away from the corners, or use seam painting.

I've run a bunch of tests (about 30 at this point), and have learned a few things, but I need to re-do a bunch of defect measurements on existing models, and possibly re-print several tests.  Several of the default settings were making the flaws much worse.  The "Star" infil at 20% was increasing the shrinkage a LOT.  Rectilinear or honeycomb work much better, and increasing the infill percentage also helps.  I saw no improvement using "external perimeters first".

colin-small, 2025-01-02:
I just discovered this thread and I have been trying to figure this issue out for a long time. Forgive me as I may have overlooked something, but how is it not related to the slicer if it doesn't seem to happen in slicers like Simplfy3D? I experience it on my Qidi X Max 3 using Orca slicer, Prusa slicer, Qidi slicer (based on Prusa Slicer), and Bambu Studio, and Qidi Studio (based on Bambu Studio). I haven't tried Simplify3D or Cura yet. If Simplify3D doesn't seem to suffer from the issue then I would consider purchasing it. It seems like wherever a new solid layer starts the line is visible on the exterior surface across the whole side of the print. I am a little newer to 3D printing so I'm sorry if I am misunderstanding the issue. I just know I seem to suffer from it.

DaGwhite, 2025-01-02:
That's the big question.  It may not be an actual "bug" in PrusaSlicer, but there is definitely something they are not doing as well as some (most?) other slicer options.  I've been able to reduce the defect by adjusting various print parameters, but it never goes away, or even gets what I would consider acceptably small.  I've largely run out of time to run more experiments for now.  Each time I fight this, I learn a bit more.  I still have a pile of test prints & measurements I would like to write up and submit here, but it may be a little while.

One of the issues I've run into is that some of the problem may be the result of the "Y-shift" bug, which is a REAL bug in Prusa Slicer ([https://github.com/prusa3d/PrusaSlicer/issues/url](url)).  If it was simply a defect resulting from the slicer settings, the orientation shouldn't matter much.  I get different amounts of layer shift depending on which may the model is placed.  I have to double check, but I believe the amount of shift I've been seeing may be a function of the Y position on the bed, which is how the Y-shift bug behaves.  It's going to be very difficult to sort things out if there are two competing issues going on.

As far as Simplify 3D goes, it may not be perfect, but every time I HAVE to get a reasonably good print of a model that exhibits this behavior in Prusa Slicer, I can go back to S3D (Version 4.7).  I get an instant significant improvement, usually enough to get the job done.  I also have S3D version 5.x, but they changed the entire user interface, and I have been devoting my time to learning Prusa Slicer while they get the bugs out out of Version 5.  It should be pretty stable by now.  Others have reported that they don't see the issue with Cura.  The "bulge" postings on the Prusa Slicer forum have more info on what slicers people have tried.

DaGwhite, 2025-01-04:
I did two final test prints before I have to go off to deal with other stuff.  It actually took several tries, but I now have "matched"  test prints from Simplify3D 4.1.2 and one from PrusaSlicer 2.9.0.   I started with the best settings I've found using PS, and copied them over to S3D.  I went through and verified as much as possible that I had duplicated the following settings:

External perimeters first
Layer height = 0.1 mm
Extruder & bed temperatures
Speeds (as best I could, the way they are specified is different in some ways)
Infill pattern (rectilinear)
Infill density (30%)
Cooling (off)
Retraction & Extra restart distance
Extrusion multiplier
Layers & Perimeters all set to 3

The two prints are significantly better than when I started this effort with a lot of the PrusaSlicer default settings.  The S3D print has slight defects on one end, and almost no defects visible on the other.   The PrusaSlicer print is not terrible, but it is noticeably worse than S3D on both ends

[image]

[image]

In all cases, the Simplify 3D defects are shallower, and less abrupt than what I get with PrusaSlicer.  I measured the width of the defects as best I could, and they varied from 0.09 mm to 0.12 mm on the PrusaSlicer print.  On the Simplify3D print, I got 0.02 mm on the good end, and 0.07 on the other end. 

Based on what I've seen, the defects can be reduced further in both slicers by increasing the infill to 50% or more, but that has other issues (like time & filament).

It may not be a "bug" per se, but PrusaSlicer has consistently produced inferior prints compared to my 6 year old copy of Simplify3D when I have models that trigger this issue.  The problem has been bad enough on at least a dozen different prints since I started using PrusaSlicer about 5 years ago that I have had to switch back to Simplify3D to get acceptable results.  

In the meantime, I'm going to toss all my test pieces and get back to work...

[image]

closed 2026-09-29 by prusaslicer-bot[bot]