prusa3d/PrusaSlicer #13194: Prusa Slicer 2.8.0 on Linux Mint 22

loops21, 2024-08-02:
### Description of the bug

Hello !

I've updated my linux mint 21.3 to 22.0. Prusa Slicer was running perfectly before the update.
When Linux updated, i've this error when I try to run Prusa Slicer 2.8.0 : 

loops@loops-LIFEBOOK-A557:~/Applications/PrusaSlicer$ ./PrusaSlicer-2.8.0.AppImage 
(process:2211): libsoup-ERROR **: 12:05:47.298: libsoup2 symbols detected. Using libsoup2 and libsoup3 in the same process is not supported.
Trappe pour point d'arrêt et de trace (core dumped)

Do you know how to correct this error ?
(Prusa Slicer 2.7.4 work correctly on my 22.0 Linux mint)

Thank you for your help !

JC

Os Infos : 

System:
  Kernel: 6.8.0-39-generic arch: x86_64 bits: 64 compiler: gcc v: 13.2.0 clocksource: tsc
  Desktop: Cinnamon v: 6.2.7 tk: GTK v: 3.24.41 wm: Muffin v: 6.2.0 vt: 7 dm: LightDM v: 1.30.0
    Distro: Linux Mint 22 Wilma base: Ubuntu 24.04 noble
Machine:
  Type: Laptop System: FUJITSU product: LIFEBOOK A557 v: N/A serial: <superuser required> Chassis:
    type: 10 serial: <superuser required>
  Mobo: FUJITSU model: FJNBB50 v: 716944-01R6901464 serial: <superuser required>
    uuid: <superuser required> UEFI: FUJITSU // Insyde v: Version 1.09 date: 06/06/2017
CPU:
  Info: dual core model: Intel Core i5-7200U bits: 64 type: MT MCP smt: enabled
    arch: Amber/Kaby Lake note: check rev: 9 cache: L1: 128 KiB L2: 512 KiB L3: 3 MiB
  Speed (MHz): avg: 600 min/max: 400/3100 cores: 1: 600 2: 600 3: 600 4: 600 bogomips: 21599
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel HD Graphics 620 vendor: Fujitsu driver: i915 v: kernel arch: Gen-9.5 ports:
    active: eDP-1 empty: DP-1,HDMI-A-1 bus-ID: 00:02.0 chip-ID: 8086:5916 class-ID: 0300
  Device-2: Chicony FJ Camera driver: uvcvideo type: USB rev: 2.0 speed: 480 Mb/s lanes: 1
    bus-ID: 1-7:4 chip-ID: 04f2:b413 class-ID: 0e02
  Display: x11 server: X.Org v: 21.1.11 with: Xwayland v: 23.2.6 driver: X: loaded: modesetting
    unloaded: fbdev,vesa dri: iris gpu: i915 display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1366x768 s-dpi: 96 s-size: 361x203mm (14.21x7.99") s-diag: 414mm (16.31")
  Monitor-1: eDP-1 model: BOE Display 0x061e res: 1366x768 hz: 60 dpi: 112
    size: 309x173mm (12.17x6.81") diag: 389mm (15.3") modes: 1366x768
  API: EGL v: 1.5 hw: drv: intel iris platforms: device: 0 drv: iris device: 1 drv: swrast
    surfaceless: drv: iris x11: drv: iris inactive: gbm,wayland
  API: OpenGL v: 4.6 compat-v: 4.5 vendor: intel mesa v: 24.0.9-0ubuntu0.1 glx-v: 1.4
    direct-render: yes renderer: Mesa Intel HD Graphics 620 (KBL GT2) device-ID: 8086:5916
Info:
  Memory: total: 4 GiB available: 3.72 GiB used: 2.14 GiB (57.6%)
  Processes: 238 Power: uptime: 8h 32m states: freeze,mem,disk suspend: deep wakeups: 0
    hibernate: platform Init: systemd v: 255 target: graphical (5) default: graphical
  Compilers: gcc: 13.2.0 alt: 9/11/12 Client: Cinnamon v: 6.2.7 inxi: 3.3.34

### Project file & How to reproduce

Just launch Appimage from command line on linuxMint 22.0
If launched from GUI, nothing append.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Linux Mint 22.0 

### Printer model

SideWinder X1

oerkel47, 2024-08-04:
I am having problems as well after upgrade to 22. 

"error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory"

loops21, 2024-08-04:
To correct this error you can execute this commands : 
`sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
sudo ln -sf /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18`

Thank you for sharing that there is a version of Prusaslicer for Ubuntu 24.04. it work fine on my computer !
This version is downloadable here : https://github.com/prusa3d/PrusaSlicer/releases

markusdd, 2024-08-04:
just a hint guys: symlinking a new major version of a library to an older one that most likely will have API incompatibilities compared to the previous version is asking for trouble.
It might work for prusa slicer but you can cause major system instabilities with stuff like this.

nazar-pc, 2024-08-07:
Doesn't start on Ubuntu 24.04 either, but please, don't mess with system libraries and `sudo` unless you REALLY know what you're doing.

Maybe duplicate of https://github.com/prusa3d/PrusaSlicer/issues/12931?

nazar-pc, 2024-08-07:
UPD: There is a separate build for Ubuntu 24.04, try it on Mint 22

loops21, 2024-08-07:
Yes this build works fine on Mint 22. 
I've removed the simlinks !

Thank's all for the help !

nazar-pc, 2024-08-07:
I think the issue can be closed then

closed 2024-08-07 by loops21