prusa3d/PrusaSlicer #13195: Organic supports that start in mid-air

funkyduck77, 2024-08-02:
### Description of the bug

I am currently struggling with a rather strange problem. 
I have a model sliced with PrusaSlicer (Ver 2.8.0+win64) and when I want to use organic props, they sometimes start in the air. What could be the reason for this?

Supports start from layer 7... 
[image]

...and become more the more layers are added.
[image]

Even rotating or repositioning the model did not change anything.

### Project file & How to reproduce

[Body_sign.zip](https://github.com/user-attachments/files/16474556/Body_sign.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

Windows 11

### Printer model

Prusa MK3.5 & MMU3

funkyduck77, 2024-08-02:
After a few tests, the error does not seem to exist if you use ‘Automatic painting’ in the support. 
[image]
[image]

chmuc01, 2024-08-14:
I have the same problem. It seems that if the organic support trees are not high (in my case 6mm) then the 'inner' branches of the trees begin in the air, meaning that support is generated from 0.6 to 1.2 mm above the initial layer.
This means there is no way that these supports can be printed correctly and the result is a failed print and a lot of material detached.
See attached file for an example. 
[OrganicSupportError.3mf.zip](https://github.com/user-attachments/files/16618718/OrganicSupportError.3mf.zip)

Jan-Soustruznik, 2024-09-04:
Hello, Thank you for your report. Same problem is already reported. I closed issue as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9601

closed 2024-09-04 by Jan-Soustruznik