prusa3d/PrusaSlicer #13196: Huge wipe tower with variable layer height

unkn-one, 2024-08-03:
### Description of the bug

When using **Variable layer height**, slicer uses current layer height instead of the layer height specified in the print profile (0.2 structural etc) when wiping, generating huge wipe tower, even though **No sparse layers** is selected.

[image]

I would expect the wipe tower to be much smaller when using **No sparse layers**, since there is no point in in wiping with the current (much smaller) layer height at a completely different Z location.

### Project file & How to reproduce

[Project zip](https://github.com/user-attachments/files/16476810/var-height-wipe.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 10

### Printer model

Prusa MK4

u89djt, 2024-08-03:
(intrigued fellow user) That's really interesting. I suppose the reason to use the layer height from the object is to ensure that the wipe tower isn't too tall for the print volume. If you have a lot of very fine layer heights in a tall object and use no sparse layers with the default height, the wipe tower could easily be taller than the object or even the print volume depending on how many layers have tool changes. It says it's experimental, so I imagine there's some playing-it-safe involved at this stage in development.
[image]
You'd have to know how many layers involve tool changes to be able to compute a sensible standard wipe tower layer height, presumably capped at the project default.

unkn-one, 2024-08-03:
Yeah, I was also thinking about that - you could use the average from all wipe layer heights for wipe tower instead.
That way you'd always have the same XY space required for all wipe moves and the wipe tower would be no higher than the actual print.

closed 2026-09-29 by prusaslicer-bot[bot]