prusa3d/PrusaSlicer #13203: Instant Crash when Slicing Large Objects

LuckyResistor, 2024-08-04:
### Description of the bug

When slicing large objects, PrusaSlicer crashes in some cases at the end of the slicing process.

### Project file & How to reproduce

## How to Reproduce

- Create a new project.
- Select the "Original Prusa XL - 5T Input Shaper 0.4 nozzle" profile.
- Select the "0.20mm STRUCTURAL" print settings.
- Add a new box to the printer bed, with the dimensions 350x350x350mm
- In the print settings, change the infill from "Grid" to "Gyroid".
- In "Multiple Extruders" disable the wipe tower.
- Press "Slice now"

### Expected Behaviour

- The box is sliced.

### Actual Behaviour

- PrusaSlicer crashed at the end of the slicing process.

## Screenshot

[image]

## Project File

[prusa-slicer-crash.3mf.zip](https://github.com/user-attachments/files/16487546/prusa-slicer-crash.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+MacOS-arm64

### Operating system

macOS 14.6

### Printer model

PrusaXL

grimthaw, 2024-08-06:
Flatpak version of PrusaSlicer v2.8.0 also crashing at 89% through slicing this [model.](https://1drv.ms/u/s!AkgkLhmowV33hMQizHHRPq7rwC6oxA?e=HpZQIV) . No error message.

LuckyResistor, 2024-08-06:
@grimthaw This does not crash PrusaSlicer on my computer. So, it depends on the computer, OS and probably the amount of memory you have. Please add your operating system and exact PrusaSlicer version to your comment. Probably the amount of RAM would also help.

closed 2026-09-29 by prusaslicer-bot[bot]