prusa3d/PrusaSlicer #13204: Negative Z offset works with numpad but not -_ key

hackwin, 2024-08-04:
### Description of the bug

Negative Z offset is sometimes entered to get the print to stick better to the bed.  In Prusaslicer, a negative value can only be entered using the "-" key on the numpad.  The "-_" key to the right of the "0)" key does not work in the field.

### Project file & How to reproduce

Printers -> General -> Z offset: 
Enter a negative value using the "-_" key and it doesn't work.  The "-" key on the numpad does work.  Both keys should work and do the same thing.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Windows 10 Pro

### Printer model

Mk4

neophyl, 2024-08-04:
Please do a search before opening new issues.  This has been reported many multiple times.

Jan-Soustruznik, 2024-09-04:
Closing in favor of https://github.com/prusa3d/PrusaSlicer/issues/13043, which contains a list of all the duplicates.

closed 2024-09-04 by Jan-Soustruznik