prusa3d/PrusaSlicer #13208: v2.8.0 - Cold extrusion - gcode generation doesnt pre-heat nozzle for "First Layer" at all

PhoenixCZE, 2024-08-05:
### Description of the bug

extruder is not heated at all, first heating is after first layer

### Project file & How to reproduce

[image]
[bowdenleft NEW.zip](https://github.com/user-attachments/files/16495079/bowdenleft.NEW.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

win64 2.7.x - 2.8.0

### Operating system

windows 10

### Printer model

Anet A8, but it doesnt matter since issue is missing gcode setting

neophyl, 2024-08-05:
The devs need your saved project file from PS.  Not gcode.  File>Save project as.  That will save a PS 3mf format file.  They need the settings from your profiles which the project will contain.  In this case the Printer profile most likely.

Given the gcode file shows that your start gcode doesn't have any lines telling it to preheat to the first layer temp,  so do you by any chance have the Printer Settings>Custom Gcode> Emit temperature commands automatically = unticked as an option ?

[image]

If that is unticked you have to explicitly add the correct lines to the start gcode.  If its ticked then the slicer should automatically add them for you.  Not always in the place you want them which is why its best to add them yourself anyway.

With the project file it would be easy to check.

PhoenixCZE, 2024-08-05:
Hi,

thanks for fast response, the setting is ticked so it should be added as
you mentioned.
[image: image.png]

3mf file is attached below.

Thanks for help

PhoenixCZE, 2024-08-05:
[temp settings plus 3mf.zip](https://github.com/user-attachments/files/16498718/temp.settings.plus.3mf.zip)

neophyl, 2024-08-05:
Thanks for that (not a dev btw - just another user)

Looking at your profile you are sort of supplying the heat extruder commands which is why its not automatically adding them.  However you are supplying ANY parameters for the M109 (which is strange as you are supplying a parameter for S0 in the End gcode section.

So you have 
;M109 ; wait for temp 
but what temp ?  For a fixed temp you would use M109 S210 for example to set 210 degrees.  If you want to use the first layer temp from the Filament profile you have to use a placeholder.

For PS you need to provide the proper placeholder (which can be found by hovering over a setting.  SO in this case it is first_layer_temperature. 

For a normal printer profile the correct block of lines are along this
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp

So the M109 S is followed by the appropriate placeholder.

I don't know where you got your Anet A8 profile from but it is a bit lacking.  If I recall the A8 is basically a MK3 clones isn't it ?
I would install a Mk3 printer profile and examine the contents and settings of the printer settings.

Correct the profile and it should work.

PhoenixCZE, 2024-08-05:
Hi,

well that M109 was commented out, I was testing it back then. Then I realize that I should not set it directly since some time you need a little bit different temperature etc ... so I commented that out.

Yes, its MK3 clone I think, powered by marlin FW.

I might try to backup settings and start vanilla settings, then inspect gcode if there is M104 before extrusion. It might be related with upgrade among versions ....

PhoenixCZE, 2024-09-12:
Hi, I just found the solution ... it looks like there is some twisted sense of humor behind it :D
You just have to go to Printers => Custom G-code and uncheck "Emit temperature commands automatically" then save settings.
After that you have to check it again and save settings, now it works perfectly.

[image]

hugomsmarreiros, 2024-12-21:
M109 won't be take into account after G80, why?

closed 2026-09-29 by prusaslicer-bot[bot]