prusa3d/PrusaSlicer #13209: 3mf files from Fusion randomly not being able to open

slart11, 2024-08-05:
### Description of the bug

Sometimes when exporting 3mf from Fusion, the file will not open. When I drag it into the slicer, I get an error.
When I export as STL, it is fine.

The 3mf can be opened in Cura, then I save, and then I can open in Prusa as well. But the model needs to be repaired.

### Project file & How to reproduce

Export 3mf from Fusion
Open with Prusaslicer - doesnt work
Open with Cura - works
Save with Cura
Then open with Prusa
[Your-Files.zip](https://github.com/user-attachments/files/16496370/Your-Files.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Win11

### Printer model

MK4

Jony01, 2024-08-13:
Hi,
Long story short: There is invalid object in 3mf.

PrusaSlicer say that there is an issue in file '3D/3dmodel.model' at line 13338.
[image]
On the line is defined item from list of object you want to print (object with id = 1).
[image]
But definition of this object is without triangles( '3D/3dmodel.model' at lines from 11 to 16).
[image]

At this moment PrusaSlicer do not ignor mistake like this in 3mf.
(but we are on the way to fix it)

slart11, 2024-08-13:
Great, then you are on it ^_^

closed 2026-09-29 by prusaslicer-bot[bot]