prusa3d/PrusaSlicer #13210: Infill Anchor in Air

phigjm, 2024-08-05:
### Description of the bug

Problem: 
The infill anchors are put randomly and directly into the air:
[image]
This causes Problems in some cases that the anchors aren't staying at their place. 

### Project file & How to reproduce

Starting PrusaSlicer (2.74 or 2.8.0 dosn't matter) -> Right Click -> Add Shape -> Cylinder

Slice with default parameter.  Select in Legend `Feature type` and remove `Perimeter` and `External perimeter` to see the anchor deffects.

The issues can be avoided by setting  `Length of the infill anchor:` and `Maximum length of the infill anchor` to 0.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0-win

### Operating system

Windows 11

### Printer model

-

neophyl, 2024-08-05:
Except on an actual print those are anchored to the perimeters and are what help secure the infill to the shell.  They are not floating in the air.  Its what its supposed to do.

phigjm, 2024-08-06:
Thank you. But why are the anchors randomly generated and not stacked above each other. This can lead I think to pour print quality as in our case the anchors don't stay well attached to the wall...

closed 2026-09-29 by prusaslicer-bot[bot]