prusa3d/PrusaSlicer #13211: Show coordinates in Measure tool

dzuelke, 2024-08-05:
It would be useful if the Measure tool was able to show X/Y/Z coordinates for a selection (if the selection is not a point, then the geometric center of the plane or circle).

This would make it much easier to align modifiers or simple volumes with features in existing models. For example, being able to measure an existing hole using its circumference or center point would allow precise placement of a larger negative cylindrical volume to enlarge the hole, or a matching "positive", then a smaller negative, cylindrical volume to shrink the hole.

jmsaltzman, 2025-03-03:
Most of the time, a vertex's coordinates are all that's necessary for making precise adjustments of the object or of modifiers. I try to position modifiers precisely for sacrificial bridges or to widen holes. Also very useful when cutting and reassembling models.

For feature completeness or creep, depending on your perspective, these would be great:
1. For an edge or vector, the two vertex positions would be very useful,
2. The position of a selected circle's center,
3. Object vs world space position would be useful for positioning a child object in parent space. Maybe a toggle?

Of course just having a single vertex's world space position would be a big win for me. Many thanks!

Tupson444, 2026-09-19:
good idea, this would improve positioning modifiers a lot