prusa3d/PrusaSlicer #13215: Crash slicing / generating infill on 2.8.0

wjessop, 2024-08-06:
### Description of the bug

PS crashes at exactly 45% into the "Making Infill" stage of the process. Prusa Slicer sits at 300% CPU and has to be force quit. This happens on 15% and 25% infill density, I've not tried other settings.

### Project file & How to reproduce

Reproduction: Open Prusa Slicer, add one of the attached STLs to the plate. It should slice fine. Now, place it on the face opposite the large screw threads (as in the screenshot), slicing will stop at 45% infill generation.

[image]

[STL.zip](https://github.com/user-attachments/files/16514789/STL.zip)

I can't add a project file because Prusa Slicer has locked up so I can't save one.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

M1 Macbook Pro, Sonoma 14.6

### Printer model

3.5

wjessop, 2024-08-06:
I switched to Gyroid infill from grid and it crashed at 45% again.

Jan-Soustruznik, 2024-08-09:
Hello, @wjessop , thank you for your report. Problem will be fixed in next release. Thank you, have a nice day

closed 2024-08-09 by Jan-Soustruznik