prusa3d/PrusaSlicer #13216: App crash while changing part type

rahmanshaber, 2024-08-06:
### Description of the bug

Changing the part type causes the app to crash.

[image]
```

PrusaSlicer

Version:   2.8.0
Build:     PrusaSlicer-2.8.0+flathub.org

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 6.9.10-200.fc40.x86_64 x86_64
Total RAM size [MB]: 41,831MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 555.58.02)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 3070 Ti Laptop GPU/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)

```

### Project file & How to reproduce

[parts, scrrenshot and project.zip](https://github.com/user-attachments/files/16516398/parts.scrrenshot.and.project.zip)

1. Here is project where some are used as support type.
2. now change the type of the part to anything else
3. app crashes 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.0+flathub.org

### Operating system

Fedora 40

### Printer model

MK3s

Jan-Soustruznik, 2024-09-11:
Hello, @rahmanshaber Thank you for report, but I cannot evoke the crash according to your steps with attached project file. This crash is happend every time when you change the type in the attached project?

rahmanshaber, 2024-09-13:
Yes that's correct, here i recreated it again.
[Screencast from 2024-09-13 21-11-29.webm](https://github.com/user-attachments/assets/dd3afc23-8e63-4f28-b608-5d863247222f)

Jan-Soustruznik, 2024-09-17:
Hello, @rahmanshaber , thank you for your answer. Yes, confirmed. Problem can be reproduces on our side. Crash is located only in Flatpack version. Before we find where problem is, you can use - .AppImage provided by us in release of 2.8.0(https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0) or latest release 2.8.1-rc1(https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.1-rc1). Have a nice day

lukasmatena, 2024-09-17:
This is very likely already tracked under SPE-2017.

rahmanshaber, 2024-11-23:
@Jan-Soustruznik  Hi, i have tried the alpha flatpak, and it's not crashing now when changing type.

closed 2026-09-29 by prusaslicer-bot[bot]