prusa3d/PrusaSlicer #13217: Add Settings for Estimated Startup and Cooldown Times

forReason, 2024-08-07:
**Description:**

It would be beneficial to have settings in PrusaSlicer that allow users to provide estimated startup and cooldown times in the format { hh.MM.ss }. These settings would enhance the accuracy of print time estimations by including the time needed for the printer to warm up before printing and cool down after printing.

### Use Case:
Imagine starting a print and the printer estimates it will take 15 minutes. You return after 15 minutes, but the printer is either still finishing up, or the print bed is too hot to safely remove the print. This can result in potential burns or permanently bending the still-warm plastic if you attempt to remove it too early.

### Proposed Solution:
1. **Estimated Startup Time Setting**: Allow users to input an estimated time for the printer to warm up before starting the print.
2. **Estimated Cooldown Time Setting**: Allow users to input an estimated time for the printer to cool down after the print finishes.
3. **Configuration Location**: These settings should be part of the printer configuration.
4. **Filament Overrides**: Ideally, these settings could be overridden per filament, as different temperatures and case opening requirements might affect the warm-up and cool-down times significantly.

### Benefits:
- More accurate total print time estimations.
- Better planning and time management for users.
- Enhanced safety by avoiding premature removal of prints.

### Negatives
- None apparent

### Example:
**Estimated Startup Time Setting:**
```
Estimated Startup Time: 00:03:00 (3 minutes)
```
[image]

**Estimated Cooldown Time Setting:**
```
Estimated Cooldown Time: 00:13:00 (13 minutes)
```
[image]

**Comparison of Estimated vs Actual Print Time:**
```
Slicer Estimate: 12.3 minutes
Actual Time (including warm-up and cool-down): 25.5 minutes
Slicer estimate with new calculation (Warm up + print + cool down ) (3 + 12.3 + 13): 28.3min
```
[image]

From the test results, I assume, the slicer already factors in a given warmup and cooldown time, but it seems too little.
For longer Prints, the time is more accurate, since warmup and cooldown dont play a huge role but its quite relevant when printing either small parts or with a high nozzle and high layer height.

neophyl, 2024-08-07:
Basically a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/12659  (but yours is prettier)

forReason, 2024-08-07:
I think this is an understatement

closed 2026-09-29 by prusaslicer-bot[bot]