prusa3d/PrusaSlicer #13218: Maximum Printer Size 1074mm

phigjm, 2024-08-07:
### Description of the bug

The Maximum Printer size seems to be around 1074mm. Is there a way to increase the printer size?
If I put a larger number the Slicer in Debug mode crashes with the Information `Coordinate outside allowed range`. 
[image]

Interrestingly in Release mode the max value is 1200mm and also a different error message ocurres.

This message seems to only occurs once in the source code:
[image]
 I tried to addapt the `hi` variable as well as removing the exception as well as changing the message text with no change. The Message text didn't even change after cleaning, rebuilding, ...

### Project file & How to reproduce

Add a custom printer within the wizzard process and enter a bed size of 1500mm x 1500mm. the error ocures right after leaving the first print field. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0-win

### Operating system

Windows 11

### Printer model

Custom Printer

neophyl, 2024-08-07:
Have you seen this existing open one https://github.com/prusa3d/PrusaSlicer/issues/6072 ?

phigjm, 2024-08-07:
Oh great. No I haven't seen this. Thank you.

closed 2024-08-07 by phigjm