prusa3d/PrusaSlicer #13219: Inner brim collision with infill

TheEvilRocky, 2024-08-07:
### Description of the bug

When enabling inner and outer brim the inner brim is created into location where there is infill to be printed. 

[image]

### Project file & How to reproduce

[6-ne_korpus.zip](https://github.com/user-attachments/files/16521108/6-ne_korpus.zip)

Make sure that inner and outer brim are enabled.
Slice the project and check first layer. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

Windows11 22H2 19045.4651

### Printer model

Prusa XL, MK4

u89djt, 2024-08-08:
(fellow user taking an interest) This is what I get in 2.8.0, windows 11 when I load and slice your project file
[image]
That looks ok to me. Did you definitely post the right file? The only thing I could come up with to explain the brim clash is a small upward offset of the icon parts, but I imagine we'd be seeing bridges.

TerPuc, 2024-08-19:
Hello @TheEvilRocky, thank you for you report. Same as @u89djt I'm not able to to reproduce the issue with the attached file. Is this issue still persisting?

closed 2026-09-29 by prusaslicer-bot[bot]