prusa3d/PrusaSlicer #13224: zoom indietro

marcellosteri1979, 2024-08-08:
### Description of the bug

Buongiorno a tutti, è da qualche giorno che come apro prusa slicer in automatico sposta lo zoom indietro e non c'è modo di avvicinare la vista, o meglio faccio zoom avanti con la rotellina del mouse ma appena lascio torna indietro, avete la soluzione?
grazie

### Project file & How to reproduce

[6.zip](https://github.com/user-attachments/files/16543573/6.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0 win64

### Operating system

windows 10

### Printer model

ender 3

u89djt, 2024-08-08:
When that happens to me it in any piece of software, it's usually caused by another input device plugged into my computer that I've absent-mindedly put something on top of :).

Quando questo mi accade con qualsiasi software, di solito è causato da un altro dispositivo di input collegato al mio computer che ho distrattamente coperto con qualcosa :).

marcellosteri1979, 2024-08-08:
non c'è nessun'altro dispositivo collegato al pc, poi succede solo con prusa e da qualche giorno, uso modellatori vari e questo problema non c'è

u89djt, 2024-08-08:
Then I'm stumped

allora sono perplesso

kubispe1, 2024-08-09:
That's strange. Do you encounter this issue with a specific gizmo, or is it happening everywhere in the Slicer? Can you support your claim with a video recording?

closed 2026-09-29 by prusaslicer-bot[bot]