prusa3d/PrusaSlicer #13225: Re-zipping the offline vendor bundle file results in a failed to extract error

ilhanesmail, 2024-08-08:
### Description of the bug

[image]

When re-zipping the offline vendor bundle on Windows, PrusaSlicer is unable to extract the .zip file when uploading vendor bundles from Local sources. This is important for the cases where customers need to modify profiles or add their own profiles.

### Project file & How to reproduce

[non-prusa-fff-offline (1).zip](https://github.com/user-attachments/files/16551351/non-prusa-fff-offline.1.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Windows 10

### Printer model

NA

Gunner087, 2024-08-10:
What software are you using to re-zip the bundle?

ilhanesmail, 2024-08-12:
@Gunner087 I'm using the windows built-in "Compress" option, similar to what you did on macOS. 

Thanks for the insight, I just tried running those steps with 7-zip at the "0-Store" compression level but I'm encountering the same error. I also confirmed that there is indeed no compression between the original .zip file and the re-zipped file (both are 45.2MB). 

I assume you did this on your MacOS? I'm trying this on a Windows 10 laptop.

ddavidebor, 2025-01-07:
I'm having the same issue, I tried a handful of options using 7zip installed through homebrew on an intel mac, but no luck.

rlamar, 2025-07-06:
It appears that the issue here isn't that you are using compression, 7zip, or anything else. I came here with the same issue, but I think it can be fixed by zipping the items at the `manifest.json` level instead of zipping the parent folder. This works for me even with the Win11 default zip utility.

closed 2026-09-29 by prusaslicer-bot[bot]