prusa3d/PrusaSlicer #13226: Add Part Fan modifier at layer specific.

Lilimoe, 2024-08-08:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Let Say I put infill 5% then suddenly increase it to 10% which create bridging of infill on top 5% infill. The slicer did not recognized this as bridging and the fan still running like 10% like how I'm printing PETG. I would like to modify the fan to run at 50% at that specific layer of transient from 5% to 10% infill
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Add modifying option for part cooling fan in layer height range
**Describe how it would work**
Be as descriptive as your skill level allows you.
Same like how you can modify the infill at specific level, please add another option to modify the part_cooling fan there
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
N/A
**Additional context**
Add any other context or paint drawing about the feature request here.
N/A

Snuff1eupagus, 2024-08-09:
While I think this request could use some refinement, this is a great request and has been long needed, I have learned how to control my fan speeds in Slic3r over the years. However, every time I adjust them I feel it's a very convoluted process. I would really like to be able to tell the fan to go faster here and slower there. yes I understand about ramping speeds up and down.

Tupson444, 2026-09-19:
still relevant