prusa3d/PrusaSlicer #13228: Physical printer (prusa link)

MarcoKockx, 2024-08-09:
**Is your feature request related to a problem? Please describe.**
When added a physical printer in printers you need to fill a username and password in the printer settings, but it will not remember when you select the physical printer tab

(Using Prusa XL)

**Describe the solution you'd like**
Automatic login on the tab Physical Printer

**Describe how it would work**
Select the printer tab and emededly loged in and see the printer status

**Describe alternatives you've considered**
do it still manal

**Additional context**

kubispe1, 2024-08-09:
Hi we are working on it. SPE-2246
Thanks

Jan-Soustruznik, 2024-09-19:
Hello @MarcoKockx, closed as duplicate of #12933 

Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases).

closed 2024-09-19 by Jan-Soustruznik