prusa3d/PrusaSlicer #13230: Filament presets not available for all printers even though "all" compatibility is checked

WildRikku, 2024-08-09:
### Description of the bug

I made a custom filament preset by changing the color of one of the generic presets. It said "all" in the compatible printers section but was not available to the Ender 3 S1 I added before. It's called "Test" in the project file I attached.

### Project file & How to reproduce

[test.zip](https://github.com/user-attachments/files/16567623/test.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.0 (Flatpak)

### Operating system

SteamOS

### Printer model

Ender 3 S1

neophyl, 2024-08-10:
When I load your project it opens with a Generic PLA @creality filament selected.  It temporarily installs the printer profile
The devs might need a copy of your whole config folder but hopefully one should be along to let you know what info they do need to look into this issue.

WildRikku, 2026-09-02:
> This issues has been labeled **legacy**. It means that it has been created before the release of _PrusaSlicer 3.0.0 alpha_ version.

So, I am now expected to install unstable software to test if a bug I had has been fixed without anyone closing the ticket? Or rather...

BUMP