prusa3d/PrusaSlicer #13231: Wipe tower volumetric speed exceeds physical limits

bombela, 2024-08-10:
### Description of the bug

Relevant filament settings:
[image]

Relevant print settings:
[image]

My max volumetric speed is 18mm³/s, here we observe the wipe tower going as high as 36mm³/s. My hotend maximum is 18mm³/s. At double what is physically possible, the wipe tower prints like trash, the extruder skips, grinds the filament etc.
[image]

The speed of the wipe tower appears to match the perimeter and infill speed, and is not capped by the maximum volumetric speed of the print setting.

Interestingly, if a max volumetric speed is set in the filament settings:
[image]

Then the wipe tower will be capped to the max volumetric speed of the filament:
[image]

In my example, I can work around the bug by using the filament max volumetric speed as a mean to set the maximum for the wipe tower, while the print setting max volumetric speed limits the rest.

Ideally there should be a max volumetric speed specific to the wipe tower. It appears reasonable to allow exceeding the print max volumetric speed to some reasonable amount as the wipe tower doesn't need quality, merely be printable.

What I report here as a bug is was a deliberate design decision, as reported in the closing comment on my pull request #9293.
This pull request added a wipe tower speed limit. A volumetric limit would be better (but can still be combined with a dedicated wipe tower speed limit if need be, though using the existing perimeter/infill print limits would work for me personally).

### Project file & How to reproduce

[wipe_tower_goes_brr.zip](https://github.com/user-attachments/files/16567915/wipe_tower_goes_brr.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

not relevant

### Printer model

Voron 2.4 & MK3S

TomBrodsky, 2025-10-16:
+1, still not functional in PS 2.9.3
see https://github.com/prusa3d/PrusaSlicer/pull/9293#issuecomment-3411562274

TomBrodsky, 2026-09-13:
Still valid.