prusa3d/PrusaSlicer #13234: Prusa slicer slicing objects into mid air

SanManTabor, 2024-08-10:
### Description of the bug

[image]
[image]
[image]
[image]
Its normal in the STL view but when I slice it, it becomes in mid air sometimes it leaves air in the middle of the model between layers, but its normal in the STL view

### Project file & How to reproduce

[Iron cross mid air messed up.zip](https://github.com/user-attachments/files/16568773/Iron.cross.mid.air.messed.up.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Windows 11

### Printer model

Prusa MK4 Pre assembled

neophyl, 2024-08-10:
Your model is broken.  It has tried to fix 224 issues but there are still 389 open edges.  Notice how the hole is now the solid part and the bu=it that should be there is gone ?  thats it 'fixing' things.  Basically depending on how it 'fixes' it depends on if the entire bottom is there or not.

Fix your model properly and it will slice fine.  
[image]

Here is a fixed version
[Eisernes_Kreuz_neo.zip](https://github.com/user-attachments/files/16568833/Eisernes_Kreuz_neo.zip)

SanManTabor, 2024-08-10:
Ok Thank you so much

closed 2024-08-10 by SanManTabor