prusa3d/PrusaSlicer #13235: Enable "Steeper ramp before obstacles" (printer/extruder configuration) by default - when disabled unpredictable crashes/layer shifts occures

RPospisil, 2024-08-10:
**Is your feature request related to a problem? Please describe.**
I've been suffering from unpredictable crashes and/or layer shifts on short (just on the 1st layer) or large prints (in 7th hour of 9 hours print on the same point). 
I suspected filament, extrusion temperature, USB flash disk, IS, .... but none of them was the root cause. 
After observing videos of failure I found that the issue was caused by the ramping lift - nozzle crashes to the near extrusion, so it ended either as crash or layer shift (when the phase stepping was enabled, this shift was typically ~10-20 mm)
Further it was strange because when I printed the same model a few moths ago without any issue. 

So I played with various settings of travel lift and found out that enabling "Steeper ramp before obstacles" resolves that issue

**Describe the solution you'd like**
Enable "Steeper ramp before obstacles" by default in printer/extruder configuration 

**Describe how it would work**
Just change default printer extruder configuration.

**Describe alternatives you've considered**
I enabled "Steeper ramp before obstacles" on my printer's configuration.

**Additional context**
I observed the same 'mystery' issue on forum - e.g., https://forum.prusa3d.com/forum/original-prusa-xl-tool-changer-assembly-and-first-prints-troubleshooting/massive-layer-shift-after-24-hrs-of-flawless-printing/#post-718449

codingcatgirl, 2024-08-21:
Wow, this actually fixed the issue i was having. Yeah, it would probably be good if this was the default.

MartinMajewski, 2024-09-01:
Same here. I experienced the same issue with shifted layers on the Y axis no matter the object's orientation and on all of my Mk4 printers, of which only one has the MMU3 installed, and I don't even print multi-color in this case. Then I came across this option, and it seems to help.

However, the print time increases significantly (e.g. from 3:13 to 3:40). I don't see a plausible reason for this.

closed 2026-09-29 by prusaslicer-bot[bot]