prusa3d/PrusaSlicer #13236: Version 2.8.0 starts several processes of Microsoft Edge WebView2

Wiki591, 2024-08-10:
### Description of the bug

Opening PrusaSlicer starts several processes of MS Edge WebView2. I am used to work on battery with my laptop. The cpu load of Edge WebView consumes much energy and cuts the battery life to one third, its not possible to park unneeded cores of the processor.
Went back to version 2.7.4

### Project file & How to reproduce

Just start PrusaSlicer and have a look in the taskmanager

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

 PrusaSlicer-2.8.0+win64-202406270929

### Operating system

Windows 10

### Printer model

Tevo Tornado

fe60, 2024-08-10:
About 5-10% CPU usage and a slowly but constantly growing memory consumption for doing nothing (doesn't matter, if the window is in foreground or in background).
Minimizing the application window brings the CPU usage down to zero.

With Prusa Slicer 2.7.4 it's always zero, shortly after stopping UI interactions. Switching back to 2.7.4, too.

PrusaSlicer-2.8.0+win64-202406270929
Windows 10 Home 22H2 (OS build 19045.4651)
Microsoft Edge 127.0.2651.98 (Official build) (64-bit)

fe60, 2024-08-11:
Overall 100.000 printers sold back in 2019 and shipping 9.000 new printers every month, today [1]. I don't know how many of your customers do slicing on a Windows PC and how many of them use the Slicer provided by the manufacturer of their printer (I do). The CPU usage of the Prusa Slicer 2.8.0 most likely wastes a lot of energy.

[1] https://www.prusa3d.com/page/about-us_77/

buhman, 2024-08-11:
Related: #12968

crasbe, 2024-08-15:
This is the same issue as #13127. I have the same issue on Windows 11.

fe60, 2024-08-16:
@Wiki591 I propose to

- close this issue (it's a duplicate of 13127)
- add :+1: emoji reaction to the initial post of 13127 (says your are affected too, without adding a comment developers have to scroll through)
- subscribe to issue 13127 to get notified on updates

TerPuc, 2024-10-10:
Hello @Wiki591 , thank you for your report. I will close this issue as duplicate. You can follow the webview issue development here: https://github.com/prusa3d/PrusaSlicer/issues/13127. Have a nice day!

closed 2024-10-10 by TerPuc