prusa3d/PrusaSlicer #13238: 'Wipe into object' gives non-intuitive results

myParakeet, 2024-08-10:
### Description of the bug

'Wipe into this object' does not intuitively mean: 'Allow this object to be printed with any extruder'. 

If user explicitly sets extruders for an object and checks 'Wipe into object', the user expects the object to be printed with the explicitly set extruders, but have wipes from other extruders as required. The user does not expect explicitly set extruders to be ignored. If 'Wipe into the object's infill' is unchecked, the user does not expect wipes to the objects infill. If this is the intended behavior, 'Wipe into the object's infill' should become checked and greyed out upon checking 'Wipe into this object'. The user does not expect disabling the wipe tower to disable wiping into an object. 

### Project file & How to reproduce

[image]
[Shape-Cylinder.zip](https://github.com/user-attachments/files/16570935/Shape-Cylinder.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Windows 10

### Printer model

Prusa XL 5 tool

kurtgluck, 2025-04-10:
This seems to be a bug to me.   I have tried Wipe into object AND/OR wipe into infill with 4 objects with those off and 2 objects with them on.  In all cases the 2 objects that you would expect to be used for wipeing slice identically to the normal ones UNLESS the wipe tower is turned on.  BUT if the wipe tower is turned on that eliminates the reason I would want to use the feature.   I think there is a problem here whereby the wipe tower flag is turning off ALL wipeing.

kurtgluck, 2026-09-18:
This is still an issue with 3.0.  Just because INDX doesnt have this issue does not mean that users of the PRUSA XL and for that matter the PRUSA XL+ will se ethis issue.   Current wipe support on the XL series of printers has lots of issues.