prusa3d/PrusaSlicer #13244: PrusaSlicer crashes when slicing a STL model

hunterkiller11, 2024-08-13:
### Description of the bug

PrusaSlicer crashes when trying to slice a specific model in specific scaling under a specific preset

### Project file & How to reproduce

Trying to slice this model: https://www.thingiverse.com/thing:2198600/files
Using 40% scaling (100% will no crash), using the built-in presets for 0.20mm QUALITY and Overture PLA for filament.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1+win64

### Operating system

Windows 10

### Printer model

Mk3s+

Jony01, 2024-08-13:
In model is 142 opened edges.
[image]
Use right button on model -> in opened context menu select "Fix by windows repair algorithm"
[image]

Already fixed model:
[Soulstone_20_exp_02_fixed.zip](https://github.com/user-attachments/files/16598519/Soulstone_20_exp_02_fixed.zip)

closed 2024-08-13 by Jony01

hunterkiller11, 2024-08-13:
Thanks for the fix and the effort. My suggestion is that the software should be stable and not abruptly crash even if the model itself is problematic.

jmd73, 2024-08-14:
@hunterkiller11 I can't get it to crash with PrusaSlicer 2.7.4 or 2.8.0. So as I see it, the crashing is fixed.

Jony01, 2024-08-14:
> Thanks for the fix and the effort. My suggestion is that the software should be stable and not abruptly crash even if the model itself is problematic.

I agree with you. 
But acheiving of this is not as easy as you can imagine. 
(but when you have idea how to do it, feel free to create pull request)