prusa3d/PrusaSlicer #13246: Configuration Wizard crashing

Bakief04, 2024-08-13:
### Description of the bug

When trying to run the configuration Wizard to add a new printer, prusa slicer crashes. 

### Project file & How to reproduce

Did a clean install from the site. I do not have access to the files currently

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8

### Operating system

windows 10

### Printer model

Prusa i3 mk3

Bakief04, 2024-08-16:
So I posted here because I saw some people also posting on Reddit about it. I can check when I get home but if I understand the program correctly then it wasn't able to access the files for some reason. Seems to be fixed now. Thanks

TerPuc, 2024-08-19:
Hi @Bakief04, good to hear it's working for you. I close this issue then. Have a good day!