prusa3d/PrusaSlicer #13249: Extreme Lag on Ubuntu 22.04.4

CunningLogic, 2024-08-14:
### Description of the bug

The sliced preview on anything but the most simple models is laggy to the point that it cant be used or manipulated. Pre-slicing the view works fine.

Ryzen 9 5950x
128gb ram
Samsung nvme ssd
Geforce RTC 3090 TI 24GB vram

[System info](https://github.com/user-attachments/files/16615401/PrusaSlicer.txt)

file:///home/jcase/Documents/PrusaSlicer.txt

### Project file & How to reproduce

[Voron_Design_Cube_v7.zip](https://github.com/user-attachments/files/16615488/Voron_Design_Cube_v7.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929

### Operating system

Ubuntu 22.04.4

### Printer model

Voron 2.4

nvtkaszpir, 2024-08-17:
duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13076 ?

CunningLogic, 2024-08-17:
After testing every release backwards, it appears the first public 2.8.0 release is where it starts.

This issue is not seen in Orca slicer or super slicer, nor 2.7.x

fe60, 2024-09-03:
Seems to be a duplicate of #12937

closed 2026-09-29 by prusaslicer-bot[bot]