prusa3d/PrusaSlicer #13250: Adding new "print settings" presets missing

zzador, 2024-08-15:
### Description of the bug

I wanted to create additional print settings presets (smaller layer height etc.) but it seems the buttons are missing. I can only add/remove filaments and printers but not print settings. I thought about making this a feature but since this is a defacto standard in every other slicer I really think it's a bug.

### Project file & How to reproduce

Open the "Print settings" Combo-box in the Plater-Tab and unlike with Filament & Printer there is no line saying "Add/Remove print settings".

OR:

Go to the Print Settings Tab and look for an "Add preset"-Button next to the "Add preset", "Rename preset" and "Delete preset" Buttons. It seems it is missing.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 10

### Printer model

Weedo Tina 2

neophyl, 2024-08-15:
No it isnt there.  Those Add/Remove links are for starting the wizard which adds preset profiles.  They arent for saving your own modified profiles anyway.

However its available from the Print Settings tab itself.  Just like it is for Filaments and Printers.  Its normal operation to modify your profile(s) to how you want them and then to save them with a non system name (cant name profiles same as system ones).
[image]
The Disc icon lets you save the current profile.

zzador, 2024-08-15:
That green flag isn't there in my version:
[image]

neophyl, 2024-08-15:
And ?  That flag is just the compatible preset option. It has nothing to do with being able to save your own customised profiles. So you can change the layer height for example and then save your modified profile as a new one using the disc icon which you do have. 

Not sure why you don’t have the flag but if I had to guess it’s because you don’t have the setting in preferences set to show incompatible presets. That option lets you see profiles that are installed but not compatible with your currently selected printer.

TerPuc, 2024-08-19:
Dear @zzador, thank you for your report. 

I think I understand what you mean however I would like to make sure. Let me recap. You want to add your own modified preset and there is no add/remove button. Do I understand your claim correctly? 
As @neophyl kindly explained in case you would like to add your own modified preset you need to modify already existing/installed one and use the save button. This will create your own modified preset which you can name all you want and save it. The original one (installed from the wizard) will remain unchanged. This is wanted.

And to your second claim - in case you would like the flag to be shown you can enable this option in _preferences > general > show incompatible print and filament presets_ 😸  This feature hides/shows incompatible presets. However if you want to use this feature, please pay attention to compatibility of presets and make sure you always have selected the compatible profile (incompatible profile may DAMAGE your printer - that is a reason why this function is in defaultly disabled).

I hope this helps. Please let us know in case you will have an issue with creating your customized preset or any other. However I will close this issue because this is not a bug. Thank you, have a nice day and enjoy your printing! ❤️