prusa3d/PrusaSlicer #13251: Extruder selector control becomes unresponsive

lcemoore, 2024-08-15:
### Description of the bug

When assigning extruders for a multi material print, the extruder dropdown control expands and then becomes unresponsive.  This occurs randomly and following numerous control events, such as dropdown, key up, or control exit. 

Usually when this happens the control appears to have focus, until you close the application. Sometimes you can re-enter the control and select a different extruder, but usually not.  Similarly sometimes I can continue to select extruders for other objects, but again usually not. 

### Project file & How to reproduce

[Unresponsive control.zip](https://github.com/user-attachments/files/16624840/Unresponsive.control.zip)

I have tried to reproduce this behavior, but so far it seems completely random. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64,  but it was an phenomena on previous versions as well

### Operating system

Win 11

### Printer model

MK4

NielschB, 2024-08-21:
I have experienced the same behavior, not directly reproduceable, random effect. I have noticed this in quite some old versions of prusaslicer already. When it becomes unresponsive, the dropdown frame looks like the extruder has not been selected yet. I'll try to capture it when I run into it again.

lcemoore, 2024-10-28:
New example with screen shots.
For this example, extruder 3 was selected, and the slicer generated gcode loading from extruder 1. 
With the slicer in this condition, you can try to change the extruder and re-slice the file, but it always generates gcode using extruder 1, the dropdown box also stays active.
I haven’t looked at the source code, but in my programming experience it seems like the dropdown box is not completing the on-change event code. 

[slicerproblem.pdf](https://github.com/user-attachments/files/17541951/slicerproblem.pdf)

closed 2026-09-29 by prusaslicer-bot[bot]