prusa3d/PrusaSlicer #13252: Slicing for MMU3 not respecting multimaterial painting designations

jack-edelberg, 2024-08-16:
### Description of the bug

I posted this problem [in the forums](https://forum.prusa3d.com/forum/prusaslicer/slicing-for-mmu3-not-respecting-multimaterial-painting-designations/), and [BaconFase](https://forum.prusa3d.com/forum/profile/baconfase/) suggested I share it here.

I'm running a Prusa MK4/MMU3 and attempting to use the multimaterial paining tool to assign colors to different parts of my model. While most of the color designations are respected, I have one surface where the slicer either uses the wrong color or uses the right color but projects the top layer perimeters onto the bottom layer.

Here is an image of the desired coloration in the 3D editor:
[image]
And here is the gcode preview:
[image]

It seems that when I assign a color, C1, to a surface, S1, that does not have another surface, S2, of the same color, C1, adjacent to S1, the slicer instead chooses a different color, C2,  for S1. C2 is always one of the colors that is present on surfaces adjacent to S1 (though admittedly, my model is not a great way to check this behavior). When S1 does have an adjacent S2 assigned C1, the slicer correctly assigns S1 color C1. BUT, as BaconFase points out in [his reply](https://forum.prusa3d.com/forum/prusaslicer/slicing-for-mmu3-not-respecting-multimaterial-painting-designations/), when this happens, the top layer perimeters appear to be projected down onto the bottom layer (see the three following images).

3D editor view (color assignments):
[image]
gcode preview (top view):
[image]
gcode preview (bottom view):
[image]

I'm not sure what to make of this, but I agree with BaconFase that this seems most likely to be a PS problem. Please let me know if you need any additional information!

### Project file & How to reproduce

[240815-prusa-slicer-multimaterial-painting-problem-20240816T012620Z-001.zip](https://github.com/user-attachments/files/16631778/240815-prusa-slicer-multimaterial-painting-problem-20240816T012620Z-001.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

Windows 11 Home version 10.0.22631 Build 22631

### Printer model

Prusa Mk4 with MMU3

TerPuc, 2024-08-16:
Hello @jack-edelberg, thank you for reporting this. We will look at it.

Issue created: SPE 2446

lukasmatena, 2024-11-21:
Fixed in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena