prusa3d/PrusaSlicer #13253: QR code modifier

reinux, 2024-08-16:
**Is your feature request related to a problem? Please describe.**
I want to get into the habit of marking URLs on my stuff so I know where to get the models next time I want one.

URLs are fine when there's enough horizontal space, but sometimes I only have a square surface to work with, which is when QR codes would come in handy.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
The ability to conveniently add a modifier, like text or cube or svg etc., but with a QR code.

**Describe how it would work**
[This QR code generator library](https://github.com/nayuki/QR-Code-generator) may be appropriate, as it supports SVG output and offers a C+++ implementation.

**Describe alternatives you've considered**
PrusaSlicer doesn't appear to support dragging & dropping images from a browser (at least on my Linux setup), so I need to first generate a QR code using, e.g. [duckduckgo](https://duckduckgo.com/?q=qr+https%3A%2F%2Fprusa.io&atb=v416-1&ia=answer), save it, and then import it.

rkiddy, 2024-08-16:
It would be kind of sweet to just be able to click to put a QR code onto, for example, the flat bottom of a build.

malowz, 2024-08-16:
i generate my qrcode online, download bmp, convert to svg with potrace, import, set as modifier for color.

u89djt, 2024-08-17:
That's a great idea! @malowz the first google hit for "qr code scg" gives me zipped svg files directly. No idea what limits there might be.

closed 2026-09-29 by prusaslicer-bot[bot]