prusa3d/PrusaSlicer #13254: svg negative volume not working

shad131, 2024-08-16:
### Description of the bug

I was trying to add an svg negative volume to one of my prints and it would not upload as a negative volume. i have done this in the past but that was with an older version of prusa slicer. 

### Project file & How to reproduce

[test fail.zip](https://github.com/user-attachments/files/16638205/test.fail.zip)
this is the file i am having troubles with. Any svg i have tried does not seem to work as a part or negative volume.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

windows 11

### Printer model

xl 5 tool

neophyl, 2024-08-16:
Please see https://github.com/prusa3d/PrusaSlicer/issues/12915  if the workaround mentioned can help

TerPuc, 2024-08-19:
Hi @shad131, thank you for your report. This bug has been already reported. Therefore I will close this issue as duplicate. The fix will be presented in the next release :)

closed 2024-08-19 by TerPuc