prusa3d/PrusaSlicer #13255: Variable temperature with variable flowrate

otaolafranc, 2024-08-16:
**Is your feature request related to a problem? Please describe.**
with high flow nozzles we are seeing different appearances in the surface finish of 3D printed parts. specially for small sections. there are some hardware solutions but less from the software part

**Describe the solution you'd like**
I just came across this video https://www.youtube.com/watch?v=P6Y8uUPd3yg talking about a post script to do exactly this, change the temperature as the flowrate changes, something like pressure advance but for temperature. the results are quite good, and is the type of solutions that prusa likes, squishing up to last drop to current hardware and allow us to get better prints of our current machines. The script is located here over an MIT licence https://github.com/sb53systems/G-Code-Flow-Temperature-Controller

**Describe how it would work**
the temperature will vary as the flowrate changes, so the filament does not spend too much time inside the nozzle.

**Additional context**
It could be also applied to improve bridges of prints at the same time.

MrBlackmidi, 2024-08-19:
hoping that this will be considered

MrBlackmidi, 2024-09-26:
@lukasmatena will this be considered or not as this would be huge especially for high speed and high strength 3d printing?

closed 2026-09-02 by otaolafranc