prusa3d/PrusaSlicer #13260: Missing default Filament types for MK4 + MMU3

techis, 2024-08-17:
### Description of the bug

I was hoping to use the MMU3 on my newly upgraded Mk4 only to find that I am missing a bunch of filament presets in PrusaSlicer for the MMU3. 
MMU3 Filaments:
[image]

Mk4 No MMU3 Filaments:
[image]

This seems to be a silly distinction to use to limit filament types. I commonly use Prusa ASA as my filament profile but one does not exist for the MMU. I also want to use my MMU to have different kinds of filament loaded for testing various kinds of prints. Without these settings available, it defeats the purpose of using the MMU3 for me. These filaments were present for the Mk3S+. 

I suppose I could use the hack suggested in [https://github.com/prusa3d/PrusaSlicer/issues/12634](https://github.com/prusa3d/PrusaSlicer/issues/12634) but it would be really nice to have these supported by default. 

### Project file & How to reproduce

Try to select a material that is listed in All such as ASA while using the Mk4 + MMU3 printer setting. 

This isn't going to be file specific so uploading a project file in this case is not necessary. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0 Win x64

### Operating system

Win 10

### Printer model

Mk4

dpackham, 2024-12-24:
Same on 2.9.0 MacOS.  cant select PC or ASA filamants

rtyr, 2025-02-06:
MMU3 (MK4/MK3.9) support for selected ABS, ASA, BVOH and PC blend filaments was just added via configuration update 2.1.10. Until this update, only PLA and PETG filaments were officially supported.

closed 2025-02-06 by rtyr