prusa3d/PrusaSlicer #13269: Retry button

3dprinting-source, 2024-08-19:
Please add a retry button to the "Physical Printer" tab incase it fails to connect.  Currently I have to restart the program to retry again.

Jan-Soustruznik, 2024-09-04:
Hello @3dprinting-source , could you more specify our request. Why you need to restart the program? Thank you

3dprinting-source, 2024-09-04:
In the "Physical Printer" tab, if you try to access a printer or octoprint that is not turned on, you get a "Connection failed" error.  The only way to clear this error and try again is to restart the prusa slicer.  Having a "Retry" button
[image]
 would be a better way to clear the error.

Jan-Soustruznik, 2024-09-06:
Hello, @3dprinting-source  , Thank you for your feedback, We created a feature request in our internal tracking. We check the possibilities how to solve this cases. SPE-2470

Lukrative525, 2024-09-19:
Came here to ask for this. Printer was booting while setting things up in PrusaSlicer, and connection failed as a result. Not a huge deal, since I was able to send the file to the printer and use the web interface in Google Chrome like I used to. Didn't want to close PrusaSlicer just to have the web interface available in the app.

oden72, 2024-09-25:
In addition, if the computer (in this case a Mac) running Prusaslicer goes to sleep during a long print, the interface to Physical Printer may become unresponsive when re-awakened. A reload button would resolve this issue.

fealXX, 2024-10-22:
This is a common issue with octoprint and prusaslicer for me aswell. :) A reload function would be highly appreciated.

IrrerPolterer, 2024-11-23:
I want to second this. This happens to me a lot and it's really annoying.

Side note: I found that, once the page is successfully loaded, I can reload it with `CTRL + SHIFT + R`. However, that does not work, if the page is in an error state and shows the `Connection Failed` message.

Lastly, I want to add that I think it's fantastic that you added the 'physical printer' feature in the first place. Having reload functionality would be the cherry on top!

Jan-Soustruznik, 2024-11-25:
Hello @3dprinting-source, @Lukrative525, @oden72, @fealXX, and @IrrerPolterer,

The "Retry" button has been added in the latest Alpha release (2.9.0-Alpha1).
You can find it here: [PrusaSlicer 2.9.0-Alpha1 Release](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1).

Thank you for your time, and happy printing!