prusa3d/PrusaSlicer #13270: Total filament estimate in slicer summary

Orob-Maslow, 2024-08-19:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
After slicing, the user gets a legend with estimated times and filament weight for each step.  In the final line, there is a total time estimate along with the first layer estimate.  I'd like a total filament use estimate as a way to gauge if I have enough filament.  

[image]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Much like there is a total time, could you add a total weight of filament used? 

**Describe how it would work**
Be as descriptive as your skill level allows you.

Each of the times are summed to get the total time.  Sum each of the total filament weights for each of the items in the legend and display in a column to the right of the total time

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I can use a calculator and add up the numbers or estimate in my head, which is generally pretty close, so it is nice to have the numbers.

**Additional context**
Add any other context or paint drawing about the feature request here.

I'm printing some very large items and it is helpful to know based on slicing settings if I can make it fit in one roll or not.  This addition would only be a time saver for me, but it would be very useful and make infill decisions much faster or provide an estimate for when I would need to be available to intervene for the filament sensor pause when it runs out.

Orob-Maslow, 2024-08-19:
call it "estimated filament usage" right under the "Used Filament" data for each step.

neophyl, 2024-08-19:
That information is located in the sliced info panel (bottom right).
[image]

So no need to add it into the slicing preview key.

TerPuc, 2024-08-20:
Hello @Orob-Maslow, thank you for your report. The info about the filament usage is located in sidebar on the bottom right in sliced info (screenshot kindly provided by @neophyl). We are currently not planning on adding this info in legend window therefore I close this report. Have a nice day and enjoy printing! ❤️

closed 2024-08-20 by TerPuc