prusa3d/PrusaSlicer #13273: Bridge starts mid-air

eduncan911, 2024-08-20:
### Description of the bug

Detect Bridging Parameters on/off does not change (see both screenshots, one with parameters as bridges normally, and one as detect bridging).   This happens on all bridge attempts on all models.  It tries to start a bridge right in the middle of the line.

There are many issues here posting similar, but different descriptions.  None of them seem to be "hey, it prints right in the middle of the air."  So, new issue?

[image]
[image]

### Project file & How to reproduce

[Tune Filament 06 - Bridging.zip](https://github.com/user-attachments/files/16667686/Tune.Filament.06.-.Bridging.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+linux-x64-GTK3

### Operating system

PopOS 22.04 LTS

### Printer model

Custom CoreXY

eduncan911, 2024-08-20:
Figured out "the bug."  "Seam" was set to Rear.  Changed it to Nearest.

The bug is that the seam position should not be honored for bridges.  Parameters is questionable when Detect Bridging Parameters is disabled.  But right now, the bug is if it is enabled, the seam position should be ignored and instead the seam should be anchored.

Jan-Soustruznik, 2024-08-20:
Hello, @eduncan911 , thank you for your report. Yes, you have right, by the change the seam position will fix the issue, but the issue is generally caused by the extra perimeters on overhangs, that should not be used for standard bridges, but for cases where the bridges can not be anchored, because then the seam may be placed to incorrect position. I will closed this ticked as Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/12708

thank you, have a nice day

closed 2024-08-20 by Jan-Soustruznik

eduncan911, 2024-08-20:
> but the issue is generally caused by the extra perimeters on overhangs

That is incorrect.  Disabling and enabling this feature has zero effect on the reported bug above.

> I will closed this ticked as Duplicate of #12708

That is a completely different issue.  They are reporting an issue with parameters without bridging.  I have reported an issue with actual the actual Bridging, not parameters.

Here it is, with extra parameters disabled, and actual bridging starting in mid-air.  

E.g. the logic to determine the seam position is flawed for bridge functionality.

[image]

koloman360, 2024-08-22:
Another example of the same issue but not related to seam position.

[image]