prusa3d/PrusaSlicer #13276: Connect to Prusa Account behind corporate proxy

SethARhodes, 2024-08-21:
**Is your feature request related to a problem? Please describe.**
The new feature in PrusaSlicer that allows you to log in to Prusa Connect within the app does not honor the system level proxy/certificate settings and as a result will not connect. I am able to log in to connect.prusa3d.com from a web browser with no issues.

**Describe the solution you'd like**
PrusaSlicer should use the system level proxy environment variables (if present): 
 - http_proxy (http://proxyurl:port)
 - https_proxy (http://proxyurl:port)
 - HTTP_PROXY (http://proxyurl:port)
 - HTTPS_PROXY (http://proxyurl:port)
 - SSL_CERT_FILE. (Path to the self-signed certificate bundle *.pem)
This would allow PrusaSlicer to accept the corporate MITM self-signed certificates.

**Describe how it would work**
The in-app Prusa Connect should just work when behind a corporate firewall.

**Describe alternatives you've considered**
The only thing that has worked is disconnecting from the corporate VPN.

**Additional context**
N/A

TerPuc, 2024-08-22:
Hello @SethARhodes, thank you for your report. 

Issue created: SPE 2454

Jtee-cmyk, 2025-01-20:
[https://forum.prusa3d.com/forum/prusa-app-known-issues-and-troubleshooting/prusa-slicer-application-http-transient-error-port-443-time-out-at-startup/](url)

Mettbrot, 2025-02-13:
We are also facing this issue.

AndiAtom, 2025-09-09:
Same issue here

holgerschrade-sketch, 2025-10-27:
We have the same issue here

vguja, 2025-10-27:
> [https://forum.prusa3d.com/forum/prusa-app-known-issues-and-troubleshooting/prusa-slicer-application-http-transient-error-port-443-time-out-at-startup/](url)

There is nothing at this link.

vguja, 2025-10-27:
Is somebody aware of any progress in addressing/solving this issue?

brianespinosa, 2026-01-02:
> Issue created: SPE 2454

@TerPuc is there any way to track the status of this issue that was created? I cannot use PrusaSlicer for precisely this issue. There is not even a way for me to force PrusaSlicer to use a certificate by modifying the app bundle.

Mettbrot, 2026-09-22:
This is still a problem with Prusa Slicer 3.0.0 alpha12, It does not honor system Proxy settings on windows, nor is there a manual setting available in the GUI (which is very limited!)

[image]

Jan-Soustruznik, 2026-10-22:
Hello, thank you for the update. Following our new [policy,](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I am closing this feature request and converting it into a Discussion thread. This will allow us to hear what the community thinks about this feature and provide a place for others to upvote it. Have a great day.