prusa3d/PrusaSlicer #13279: Gear Bearing has connected external perimeters when arachne generator is used

jiri-jirus, 2024-08-22:
### Description of the bug

Arachne generator creates unwanted bridges between parts of "gear bearing" prusa sample object. 

### Project file & How to reproduce

Try to slice gear bearings sample stl on prusa slicer 2.8.0.
layer height 0.10 to 0.20 tried and affected by this issue with arachne. 
classic generator works ok
[image]
[image]

[Gear_Bearing.zip](https://github.com/user-attachments/files/16720038/Gear_Bearing.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 10

### Printer model

Ender 3 @ 5 drivers with Matrix Extruder

jiri-jirus, 2024-08-22:
here screenshot with arrows pointed on problematic parts
[image]

neophyl, 2024-08-23:
Change the slice gap closing radius from the default of 0.049.  I changed it to 0.005 and then the perimeters are not merged.  Seen this lots of times with gear prints and tolerance test prints, both here and the Prusa forums.  
[image]

jiri-jirus, 2024-08-23:
well, when I used just  "Split to parts" on the STL object, it stopped doing this merging mischief.

closed 2026-09-29 by prusaslicer-bot[bot]