prusa3d/PrusaSlicer #13280: Crash when G-Slicing

Zahlen0, 2024-08-23:
### Description of the bug

When I try to slice my meshes in PrusaSlicer, the program tends to crash during the slicing steps. Sometimes it will just need me to add a brim, other times i scale down the mesh and it works. I'm just wondering why this is occuring.

### Project file & How to reproduce

[File Testing.zip](https://github.com/user-attachments/files/16720711/File.Testing.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

Windows 11 Pro

### Printer model

MK3s+

Zahlen0, 2024-08-30:
Update: i figured out a workaround: run prusaslicer in power efficiency mode (setting in task manager). I'll update if there's ever a time it doesn't work.

TroelsL, 2024-09-21:
I have been experiencing much the same issues. General instability to the point of not being usable. (nearly all models will crash during Slicing).

I was very sceptical of this solution, but I have tried many other things, and it seems that this actually works.

Note: I have tried downgrading to latest 2.7, and it showed much of the same problems.

aaaaron, 2024-09-29:
Crash on almost every slice using Windows with 2.8.1; setting Efficiency mode in Task Manager did resolve the issue

tsellis74, 2024-11-24:
This seems to resolve the same issue for me as well.  Do you know if there is a way to set that permanently?  If I close out of PrusaSlicer and reopen, I have to set it back into Efficiency mode again as it doesn't automatically come up in that mode.

closed 2026-09-29 by prusaslicer-bot[bot]