prusa3d/PrusaSlicer #13281: Unsightly Seam

charris3621, 2024-08-23:
### Description of the bug

Seams are more noticeable after slicer and firmware update to latest versions of both.
[image]

### Project file & How to reproduce

[Base for craft.3mf.zip](https://github.com/user-attachments/files/16721150/Base.for.craft.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0+MacOS-x64

### Operating system

Sonoma 14.5

### Printer model

Prusa Mini

charris3621, 2024-08-23:
The model on the left is perfect. It is smooth with no seams showing. After the update (and no other changes) the seam on the model on the right is quite terrible.

u89djt, 2024-08-23:
(fellow user curious as to what could be causing it - if it's a bug, I'm no use. If it's not, I think there are things worth trying) You've changed all the acceleration parameters, which will affect behavious at stops and starts such as seams. How does it look if you use the default values?
[image]
You've increased infill overlap, which will add to any squish effects, but I don't know if that's relevant.
[image]
It's impossible to tell what's happening in the photo - too low resolution - but they look like different filaments were used. Seams can be affected by filament behaviours. Maybe it's a different filament, or have you painted the other one? If so, did you sand it?
Some things to check to start working out how to improve the outcome. I'll be interested to see what you find (and so will the Prusa people, I think!)

charris3621, 2024-08-23:
Thank you, Dave. You’re correct, they are different filaments. I used the “good” one because that’s the only model I have left that does not have the seam. When I noticed the seam, I used several different filaments with the same result. I turned on the feature in Prusaslicer that allows me to view where the seam will be located. I changed it to “rear” which makes the print kind of acceptable. These are prints that I am selling after someone saw how nice they look. The obviously visible seam may change her mind.I have not changed any settings and am using a Prusa mini with original settings. I hope Prusa fixes this in the next update.Prior to updating I had no visible seams in any of my prints. Thanks for taking time to comment.Cindy

image.png (view on web)
image.png (view on web)

u89djt, 2024-08-24:
@charris3621 you're very welcome, this is a great place to learn about potential problems before I crash into them :) The settings in the 3mf file you provided aren't the defaults, so do you mean that your previous prints were executed with those different acceleration parameters, too? Have a look at the screenshots I provided - you can see that the acceleration parameters are not the defaults expected by the slicer. Just hit the reset (circular arrow) on each. Prusa choose - and change with iterations of the slicer - the defaults to make everything work well together. Troubleshooting starts with a print that uses as many defaults as possible, especially given that the non-default settings are definitely relevant at the problem.

charris3621, 2024-08-28:
Thank you for your help. 

The model I am printing has several parts that get put together with one part sliding into another. The recent seam problem not only makes the model look terrible, but creates extra filament that I need to file off in order for it to fit properly. I am not quite understanding everything, but I opened print settings and chose .20mm SPEED @ MINI. That choice seemed to set everything to default. Is that the correct way to have all the settings be the default settings?

u89djt, 2024-08-28:
Right, if you were in a freshly installed slicer with all the default settings in the profiles and created a 3mf file without changing anything, that would give you the defaults. In your case, you've got different settings in that profile to what 2.8.0 is expecting. You'll see that it's put "(modified)" after the name of the profile for me when I open it.
First question: do you see "(modified)" at the end of the 0.2 speed profile name when you open the file like I do?

Jan-Soustruznik, 2024-09-04:
Hello, @charris3621 I also recoment to check if new Configuration were released by _Menu > Configuration > Check for Configuration Updates_

charris3621, 2024-09-04:
Thank you for your care and concern. I was at least able to move the seam to the back of the model. It still looks terrible, but at least it is not in the front. I will certainly do as you suggest and am thankful for your help.Cindy

Jan-Soustruznik, 2024-09-06:
@charris3621  I also recoment to visit our knowegle base: https://help.prusa3d.com/article/seam-position_151069

anyway I suggest to continue on Prusa Forum(https://forum.prusa3d.com/forum/original-prusa-mini/), where you can find tips how hidde the seams or how calibrate your printer and other helpfuly things about printing. 

I will close this ticked as solved, because I think that is not neccessary to leave it open. 
Thank you, have a nice day.

closed 2024-09-06 by Jan-Soustruznik