prusa3d/PrusaSlicer #13283: Feature Request: Print 2nd Perimeter First Option

FurrySphere, 2024-08-23:
**Wobbling Defects Related to Irregularities of Z Axis Movement**
It's almost impossible to make Z movement so precise that there're no slight deviations in layer heigh. And the more perimeters are used the more pronounced defect it creates. It happens because each next perimeter is being slightly pushed away by previous one and the more perimeters have been printed the more shift you get.

**Current Workarounds**
There're two workarounds for 3+ perimeters slicing that have their own flaws:
1. Enable "print external perimeter first" option. It almost completely eliminates the defect but not applicable to models having complex overhangs.
2. Print with 2 perimeters and long infill anchors to effectively get robustness of 3 perimeters. The main flaw of the method that we can't have more than 3 perimeters.

**The Solution**
Add option "print 2nd perimeter first" that rearrange perimeters printing sequence in the way: 2 -> 1 -> 3 -> 4 -> ...

**How It Would Work**
It works because we're moving shifting of perimeters lines due to Z axis movement inconsistencies from outer perimeters to inner ones and therefore the outer surface has much better quality.

**Additional Context**
I've prepared testing sample images to show perimeters count / printing order influence on the defect.
[image]
[image]

eduncan911, 2025-04-13:
in Orca, we call this "Inner-Outer-Inner" for parameter type.  

Also, duplicate of #12703

Nbeknel, 2025-04-14:
> in Orca, we call this "Inner-Outer-Inner" for parameter type.

As I understand @FurrySphere 's request, it's the opposite of OrcaSlicer's "Inner-Outer-Inner":\
2 -> 1 -> 3 -> 4 -> ... (this FR)\
vs\
... -> 4 -> 3 -> 1 -> 2 (OS's "I-O-I")

FurrySphere, 2025-04-27:
> As I understand [@FurrySphere](https://github.com/FurrySphere) 's request, it's the opposite of OrcaSlicer's "Inner-Outer-Inner": 2 -> 1 -> 3 -> 4 -> ... (this FR) vs ... -> 4 -> 3 -> 1 -> 2 (OS's "I-O-I")

Exactly. Because it's important for overhangs quality to have 2nd perimiter in place when the 1st is being printed.

closed 2026-09-29 by prusaslicer-bot[bot]