prusa3d/PrusaSlicer #13284: Object modification not respecting random seam location

tbrock47, 2024-08-23:
### Description of the bug

When specifying the seam location on an object, the slicer does not respect the selection.
This can easily be duplicated by placing a cylinder object on the print bed and modifying the seam selection to random.

[image]

### Project file & How to reproduce

**Project File:**
[Shape-Cylinder.zip](https://github.com/user-attachments/files/16728739/Shape-Cylinder.zip)

**Steps to Reproduce:**
1. Open a new slicer project.
2. Insert cylinder object.
3. Modify cylinder objects to use random seams. (This assumes the slicer default is aligned.)
4. Slice the project.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Windows 22

### Printer model

Original Prusa MINI & MINI+ Input Shaper

neophyl, 2024-08-23:
Think this is duplicate of https://github.com/prusa3d/PrusaSlicer/issues/12916

tbrock47, 2024-08-23:
> Think this is duplicate of #12916

You may be correct. I didn't come across that issue when searching. It does sound similar but #12916 is lacking a bit of detail.

Jan-Soustruznik, 2024-09-04:
Hello @tbrock47 , thank you for your report. Yes that is duplicate of #12916 . Reported issue will be fixed in next release. Thank you, have a nice day. ticked is closed as duplicate.

closed 2024-09-04 by Jan-Soustruznik