prusa3d/PrusaSlicer #13285: Grey g-code

gaellafond, 2024-08-24:
### Description of the bug

When previewing the gcode after slicing, if you disable "Perimeter", "External Perimeter" and "Top solid infill", the part turns grey.
[image]

### Project file & How to reproduce

1. Import a STL in PrusaSlicer
2. Slice
3. Disable "Perimeter", "External Perimeter" and "Top solid infill"

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+linux-x64-GTK3

### Operating system

Ubuntu 24.04 LTS

### Printer model

Prusa MINI+

gaellafond, 2024-08-24:
**NOTE**: I can reproduce this bug with most STL files, by just disabling the "External Perimeter". But, with some STL, I have to disable all 3: "Perimeter", "External Perimeter" and "Top solid infill"

neophyl, 2024-08-24:
Can confirm that this also happens on Win10 machine too.  

One thing to add is that if you enable show seams (or any of the bottom options as long as the slice contains one of them) then it will display correctly.

Viproz, 2024-12-24:
This still happens in 2.9.0, not on every model but on many and it sometimes depends on the way they are oriented.

Steps to reproduce:
1. Download the stl file here https://www.printables.com/model/1241-fruit-fly-trap-for-soda-bottle/files
2. Open it in Prusaslicer 2.9.0
3. Slice it
4. Unselect "External perimeter"

[image]
[image]

gaellafond, 2026-09-07:
This issue is still present in the latest version of Prusa Slicer.

Version: 2.9.6