prusa3d/PrusaSlicer #13286: Cannot select PrusaLink for new physical printer

m1dnight, 2024-08-24:
### Description of the bug

I have the newest version of PrusaSlicer (see details below) and firmware version 6.0.4+14924 on my MK4. 

I wanted to add a new physical printer without input shaping, but I can only select PrusaConnect as the option to connect to the printer. PrusaLink is missing. 

When I try to add a new physical printer with the input shaper profile, I do have the option to select PrusaLink to connect the printer. 

See the attached screenshots.

[image]
[image]

PrusaSlicer 

```text 
PrusaSlicer
Version:   2.8.0+MacOS-arm64
Build:     PrusaSlicer-2.8.0+MacOS-arm64-202406270929

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 14.1 (Build 23B74)
Total RAM size [MB]: 34,360MB
OpenGL installation
GL version:   4.1.0 (4.1 Metal - 88)
Profile:      Core
Vendor:       Apple
Renderer:     Apple M1 Pro
GLSL version: 4.10.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_APPLE_client_storage
GL_APPLE_container_object_shareable
GL_APPLE_flush_render
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_texture_range
GL_ARB_ES2_compatibility
GL_ARB_blend_func_extended
GL_ARB_draw_buffers_blend
GL_ARB_draw_indirect
GL_ARB_explicit_attrib_location
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_occlusion_query2
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_subroutine
GL_ARB_shading_language_include
GL_ARB_tessellation_shader
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_cube_map_array
GL_ARB_texture_gather
GL_ARB_texture_query_lod
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_sRGB_decode
GL_NV_texture_barrier
</details>
```

### Project file & How to reproduce

Not relevant I think.
[Prusa.3mf.zip](https://github.com/user-attachments/files/16736674/Prusa.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+MacOS-arm64

### Operating system

macOS Version 14.1 (Build 23B74)

### Printer model

Prusa MK4

michalk-k, 2024-08-25:
The same for MINI.
Moreover, when selecting OctoPrint for non-IS profile, the Slicer
* behaves during the test like it was PrusaConnect 
* After saving the physical printer, it behaves like it was PrusaConnect selected (luckily it prints through octoPrint properly)
* after entering the edit of the physical printer, the chosen HostType is PrusaConnect (instead of OctoPrint).

VinceKezel, 2024-09-01:
Same for me, PrusaLink no longer shows up in the host type dropdown. (see image.) I have to export gcode to desktop then upload+print from the Prusa Connect tab in prusaslicer. Selecting host type PrusaConnect always gives me a forbidden error, but thats a different issue.

[image]

PrusaSlicer2.8.0+win64
Prusa Mini+ 6.1.2+7894

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

closed 2024-09-12 by lukasmatena