prusa3d/PrusaSlicer #13287: Blank screen in Plater view

macsddau, 2024-08-24:
### Description of the bug

Nothing is displayed in in plater view; no printer bed, no tool bar (see picture). Drop any stl file, slice and crash.
[image]

Linux 5.15.0-113-generic #123-Ubuntu SMP Mon Jun 10 08:16:17 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
NVIDIA TU117GL [T1000 8GB] driver: nvidia v: 535.183.01 pcie: speed: 8 GT/s lanes: 16
    ports: active: none off: DP-3 empty: DP-1,DP-2,DP-4 bus-ID: 01:00.0 chip-ID: 10de:1ff0
    class-ID: 0300

### Project file & How to reproduce

Only opening Prusa Slicer

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0

### Operating system

Linux 5.15.0-113-generic #123-Ubuntu SMP Mon Jun 10 08:16:17 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

### Printer model

na

bkerler, 2024-12-10:
Duplicate of #8546.

Solution: If you use Flatpak:
`flatpak uninstall PrusaSlicer`, choose all
`flatpak install PrusaSlicer`

Reason: On a graphic driver update, the package FreeDesktop might miss the proper driver and reinstalling the flatpak will ensure the right graphic driver is used.

reinux, 2025-06-04:
Having the same issue, black screen in my case. NixOS Nvidia card with driver version 570.153.02. Anyone else?

icecreammatt, 2025-06-04:
> Having the same issue, black screen in my case. NixOS Nvidia card with driver version 570.153.02. Anyone else?

I had same issue also on NixOS with Nvidia with 570.513.02. I used the Windows version with Proton as a non steam game for a while as a work around but recently switched to the Flatpak one listed above and that one works great.

reinux, 2025-06-04:
Cool, I'll do that. Appreciated!

closed 2026-09-29 by prusaslicer-bot[bot]