prusa3d/PrusaSlicer #13297: Crashing Printer When No Infill

AdamS1187, 2024-08-26:
### Description of the bug

Just updated to version PrusaSlicer 2.8.0. I was tuning my printer, Elegoo Neptune 4 Plus, and everything was going fine until I tried printing in vase mode. Every time I would upload the gcode to the printer the printer would restart. I went back and tried printing with normal settings and the print would start fine. I tested each setting individually and figured out it was the infill that was causing the problem. I then tried printing in vase mode in Cura and had no issues.

Thanks
Adam

### Project file & How to reproduce

[Shape-Box.zip](https://github.com/user-attachments/files/16743265/Shape-Box.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 10

### Printer model

Elegoo Neptune 4 Plus

gaellafond, 2024-08-28:
Did you disable "Binary GCode"? Not all printer supports that feature.

> Preferences > Other > Use binary G-code when the printer supports it

You can also disable it for your printer

> Printers > General > Advanced > Supports binary G-code

AdamS1187, 2024-08-28:
I disabled binary code but the problem is still there. Right after the printer homes I get a flash from the led lights and the printer goes into a reboot. If I turn on infill the print starts normally.Sent from Samsung Galaxy smartphone.

  #13297)

gaellafond, 2024-09-04:
That sounds like a problem with the printer. Maybe there is not enough power going to the printer. That could be an issue with the power supply. Did you try to contact Elegoo?

AdamS1187, 2024-09-04:
It's funny that it only started after updating prusaslicer and only with no infill. Printer works fine with cura. I will open a ticket with elegoo.Sent from Samsung Galaxy smartphone.

  #13297)

AdamS1187, 2024-09-04:
>   #13297)
> Hello, @AdamS1187 , I dont know if we can help you... Did you try to flash FW on Printer or check the start g-code?

The printer has the most current firmware.I will have to compare the start code with what's in cura since the printer prints normally with cura.Sent from Samsung Galaxy smartphone.

Jan-Soustruznik, 2024-09-06:
Hello, @AdamS1187 , did you find some differences?

gaellafond, 2024-09-08:
> It's funny that it only started after updating prusaslicer and only with no infill.

I have a feeling that's unrelated. The sequence of command might just be tipping the printer over its limit. It's just a coincidence that it happens with those settings.

Maybe the printer is struggling with all the comments at the beginning of the file? You can edit the GCode file and remove every lines that starts with a semi-colon ";". It's possible that the printer is trying to interpret them and it's just a bit too much for it's processor...

Jan-Soustruznik, 2024-09-09:
@gaellafond  You can disable to generate G-code thumbnail by deleting field located _Printers > General > G-code thumbnails_ if filled and then use the regular expression, I think, this should works, `^(\s)*(#|;).*(\r\n|\r|\n)?` located Print _Settings > Output options > G-code substitutions_ but before print check the g-code - I didnt test if it will works at 100% and if will be safe. So when you will try to print be near the printer for quick turn off printer to awoid some defect of printer .. Anyway this seems that this is problem with FW. I suggest to ask manufacturer for solution...

Jan-Soustruznik, 2024-09-09:
@gaellafond I dont know If we can get some solution. I suggest to ask printer manufacturer for some solution ...

gaellafond, 2024-09-10:
> @gaellafond I dont know If we can get some solution. I suggest to ask printer manufacturer for some solution ...

You are probably right, but it's worth diagnosing the issue. It would be much easier for Elegoo to fix the issue if we can say "that sequence of gocde crash the printer" rather than "my printer crash when slicing that part with those settings with this slicer...". Also, if we can isolate the bug, it might be possible to find a workaround to prevent it from happening again, until Elegoo fix it in firmware (assuming it's a bug in the firmware).

closed 2026-09-29 by prusaslicer-bot[bot]