prusa3d/PrusaSlicer #13298: Bridge settings not used with soluble support interface

flynny75, 2024-08-26:
### Description of the bug

Slicer is not using bridging settings when using a 0 (soluble) support z distance. This is problematic when using PLA support in a PETG print and painting in support areas. Areas not supported are using solid infill settings rather than bridge settings.

### Project file & How to reproduce

* Have model that does bridging
* Slice, see bridge parameters used in gcode preview
* Paint on some support
* Enable support for enforcers only
* Change support top contact Z distance to 0 (soluble)
* Enable support layer synchronisation (required)
* Slice
* Observe bridge areas are using solid infill settings

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 11

### Printer model

XL 5TH

neophyl, 2024-08-26:
Yes, that is the way its designed.  By setting the distance to zero you are turning off bridging.  This has been raised before iirc.
Ah yes, a search turns up
https://github.com/prusa3d/PrusaSlicer/issues/10939
https://github.com/prusa3d/PrusaSlicer/issues/8884
https://github.com/prusa3d/PrusaSlicer/issues/6868

As mentioned by Rtyr here https://github.com/prusa3d/PrusaSlicer/issues/6868#issuecomment-919949654  you should set a very small contact distance such as 0.01 if you want to keep PS using bridging.

flynny75, 2024-08-26:
Unfortunately that means I can't use tool 1 for the perimeter of the wipe tower, which is required for the structural integrity of the tower when using materials that do not adhere

doktormerlin, 2026-09-20:
this is an issue that still exists with PS3.0, so it should NOT be closed

SquirrelEngineering, 2026-10-31:
I was about to open a new issues when I found this one. 

When using "soluble support" with a "Top contact Z distance" of 0mm, PrusaSlicer globally disables bridge detection for the entire layer. This creates a severe limitation when a single print layer contains both a region touching soluble support and an internal hollow cavity that receives no support. 

Because bridge detection is completely disabled, the printer attempts to extrude solid infill over the empty internal cavity using standard speeds and flow ratios, and in my case, not bridging over the magnets cleanly (or for a hollow region, dropping filament into mid-air and causing print failure in the hollow zones.)

Expected Behavior / Proposed Solutions
Ideally, PrusaSlicer's geometry engine should evaluate bridge detection locally rather than globally per layer. If a region of a layer is completely unsupported by any underlying geometry or support material, it should utilize standard bridging flow, speed, and fan overrides.

If a fully automated geometry fix is too difficult to implement due to legacy code, two alternative user-facing features would solve this beautifully:

1. Automated Fix: - Localized bridging detection that applies standard bridging logic exclusively to unsupported zones on a layer, ignoring the 0mm restriction elsewhere.

2. UI/Tool Fix ("Paintable Bridging"):  A "Paintable Bridge" or "Paintable Settings Region" brush tool. Similar to paint-on supports or seams, this would allow users to manually brush a specific face or internal overhang area to force bridging behavior, fan speeds, or custom flow rates without needing to set up complex local modifier volumes.